arduino/Arduino #8850: Download for Arduino1.8.9 for Linux 64bit does not include 'install.sh'

PaulSchulz, 2019-05-04:
There is no 'install.sh' script in the Arduino 1.8.9 downloadable package for Linux 64bit.
(This may also be the case for the other variations as well, but I haven't checked.)

The Arduino Nightly package has the 'install.sh' script as appears to install correctly.There is no 'install.sh' script in the Arduino 1.8.9 downloadable package.

Works: 1.8.0 Nightly Build 2019/04/18 12:25

per1234, 2019-05-06:
I just downloaded Arduino IDE Linux 64 bits and the install.sh is there. Did you download the file from the Arduino website?:
https://www.arduino.cc/en/Main/Software

Would you mind downloading it again and double checking?

PaulSchulz, 2019-05-09:
Thank you.

I can confirm that the script in present in the Linux 64bit package I have just downloaded (1.8.9)
 (arduino-1.8.9-linux64.tar.xz)

closed 2019-05-09 by PaulSchulz