arduino/Arduino #8851: Upload fails after native USB board is disconnected w/ Serial Monitor open

per1234, 2019-05-05:
Arduino IDE Hourly Build 2019/04/18 12:33 / Arduino IDE 1.9.0-beta build 105, Windows 10 64 bit

1. Plug the USB cable of a native USB (Leonardo, MKR, Due (Native USB Port)) into your computer.
1. Tools > Serial Monitor
1. Unplug the USB cable.
1. Close Serial Monitor (to avoid https://github.com/arduino/Arduino/issues/8518 when using Arduino IDE 1.9.0-beta).
1. Plug the USB cable of the board into your computer. It doesn't matter whether this is done before or after closing Serial Monitor.
1. Sketch > Upload

The upload process completes successfully, but the Arduino IDE doesn't recognize this and after a delay the upload is shown to have failed with the error:
```
Couldn't find a Board on the selected port. Check that you have the correct port selected.  If it is correct, try pressing the board's reset button after initiating the upload.
```
You must now exit the Arduino IDE and start it again before you can upload without the spurious failure.

When using Arduino IDE 1.8.9 or the Beta Build, disconnecting the board while Serial Monitor is open and then reconnecting causes multiple duplicate ports to be shown in the **Tools > Port** menu. This does not occur with Arduino IDE 1.9.0-beta, but the spurious upload failure still does occur.

[image]

The duplicate ports and spurious upload failure issue does not occur with Arduino IDE 1.8.8.

Originally reported at:
- https://forum.arduino.cc/index.php?topic=613070
- https://forum.arduino.cc/index.php?topic=443558.msg4160634#msg4160634

facchinm, 2019-05-06:
Hi @per1234 , thanks for reporting here.
This looks like a windows-only bug since I'm unable to reproduce on Linux under any circumstance. I'll take a look later using a VM

per1234, 2019-05-06:
I just gave it a try on Linux and indeed I could not reproduce the issue. So I think you're right about it being a Windows-only thing.

GreyArea1966, 2019-07-08:
I seem to be having a similar problem with my Trinket M0...
[image]
1. It will not automatically recognise when I upload a sketch
2. IT seems every time the M0 is reset, it bunny hops to a new COM port...IDE will have the old one selected, but in the list is another port detected as having the M0
3. If I select the new Port (with the M0 attached) and compile, IDE will "hunt" the COM ports but will not detect the new one until the reset switch is pressed on the board. At this point, the sketch uploads, but if I got back into IDE, the board will have "hopped" to the other COM port...leaving this long trail behind it.

GreyArea1966, 2019-07-08:
BY the way - I have not yet used Serial Monitor with these M0 processors as I have only had them a day. This just happens whenever I connect one.

GreyArea1966, 2019-07-08:
Another example...Arduino IDE says M0 connected to COM5....but in dropdown clearly shows on COM7 (albeit multiple times)

[image]

If I try to upload now...

"processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:152)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:77)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$UploadHandler.run(Editor.java:2070)
	at java.lang.Thread.run(Thread.java:748)
Caused by: processing.app.SerialException: Error touching serial port 'COM5'.
	at processing.app.Serial.touchForCDCReset(Serial.java:107)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:136)
	... 5 more
Caused by: jssc.SerialPortException: Port name - COM5; Method name - openPort(); Exception type - Port not found.
	at jssc.SerialPort.openPort(SerialPort.java:167)
	at processing.app.Serial.touchForCDCReset(Serial.java:101)
	... 6 more
"

If I select COM7 and upload;

[image]

Sometimes that process will find the chip itself, others it will fail, or I press reset.

If I try to upload again;

[image]

Then, one more try, it uploads...but when I go back to IDE Ports list...

[image]

expecting to hit COM255 by the weekend at this rate!

GreyArea1966, 2019-07-08:
Sorry for many pictures...but I have Parkinsons and typing is difficult...

facchinm, 2019-07-08:
@per1234 @GreyArea1966 the multiple entries bug gets solved by https://github.com/arduino/Arduino/pull/8046 (also merged in Beta 107). 
About the error on upload after opening the serial port, it is due to Windows caching the previously opened port, so this code path fails
https://github.com/arduino/Arduino/blob/00a7546fb43edb46bc56c55196f439fccba971b1/arduino-core/src/cc/arduino/packages/uploaders/SerialUploader.java#L203-L216
It used to fail too before https://github.com/arduino/Arduino/commit/ffba05fd31dc266892ece93ff7d9306d9f15ad18 but didn't report any error.

I'm wondering if partially reverting that patch could be the right approach. @cmaglie ?

facchinm, 2019-07-08:
I just pushed https://github.com/arduino/Arduino/pull/8046/commits/d2f8e15d9e5cc07e478e73e7a5f20b7f81e8560b in PR #8046 to also fix the upload problem.
Can you please test them as soon as @ArduinoBot produces the builds? Thanks!

commit ef8d669f (2019-07-18): Don't report an exception if waitForUploadPort fails after successfull upload

Fixes #8851
closed 2019-07-18 by facchinm