arduino/Arduino #8853: Library Manager not Updating

vmasdani, 2019-05-05:
Greetings, I want to ask why the latest version of my library does not get updated on the Library Manager. According to arduinolibraries.info, it's still in 1.1.2, but it's actually already in 1.2.0
Here's the Library: [link](https://github.com/antaresdocumentation/antares-esp8266-http)
Here's the arduinolibraries list: [link](https://www.arduinolibraries.info/libraries/antares-esp8266-http)
Thank you

per1234, 2019-05-05:
The reason why your 1.2.0 release was not picked up by the Arduino Library Manager indexer is because the `version` value in library.properties was still set to 1.1.2 at the time the 1.2.0 release was created:
https://github.com/antaresdocumentation/antares-esp8266-http/blob/1.2.0/library.properties#L2

You need to make sure to update the `version` value in library.properties before creating a GitHub release or Git tag.

You'll need to either create a new release/tag or else redo the 1.2.0 tag.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

closed 2019-05-05 by per1234

vmasdani, 2019-05-05:
Oh thank you very much! Might as well change the version