arduino/Arduino #8856: Errore di programmazione : exit status 1 expected unqualified-id before numeric constant

patverg, 2019-05-06:
Sono alle prime armi ma ho nozioni di informatica e circuiti e ho ricontrollato più volte il circuito e il programma ma non trovo anomalie e quando lancio la verifica mi esce l'errore e si evidenzia al punto 8
1 int switchState =0;
2 void setup() {
3 pinMode (3.OUTPUT);
4 pinMode (4.OUTPUT);
5 pinMode (5.OUTPUT);
6 pinMode (2.INPUT);
7}
8 void loop() {
9  switchState = digitalRead(2);
10 // questo è un commento
11 if (switchState ==Low)  {
12 // il pulsante non è premuto
13 digitalWrite(3. HIGT); // LED verde
14 digitalWrite(4. HIGT); // LED rosso
15 digitalWrite(5. HIGT); // il pulsante non è premuto
16 }
17 else { // il pulsante non è premuto
18 digitalWrite(3. LOW); 
19 digitalWrite(4. LOW); 
20 digitalWrite(5. HIGT); 
21 delay(250)//aspetta un quarto di secondo
22 // cambia gli stati dei led
23 digitalWrite(4. HIGH); 
24 digitalWrite(5. LOW); 
25 delay(250)//aspetta un quarto di secondo
26  }
27 }// torna indietro all'inizio del loop

Grazie per la cortese risposta

per1234, 2019-05-06:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2019-05-06 by per1234

patverg, 2019-05-11:
Grazie ora provvedo