arduino/Arduino #8858: Board / Library Manager] split ridiculus long menus into sub-menus

datenheim, 2019-05-07:
The menu list of the board-manager is very (too?) long and scrolling in this menu is not convenient and seems to have (had) many bugs.

Indeed the Boards already are grouped by "Arduino AVR-Boards" etc.
These groups could/should simply be transformed to a submenu.

Thus boards are simpler and quicker to reach and scroll'n'search is reduced largely.

Same may happen with the library manager. There also a grouping is existing and could be used.
Libraries not fitting into the three groups could get an own submenu.

I have only ESP32, ESP8266 and Micro/Mini/MegaCore Boardmanagers and even on FHD+ screen the menu is 3x longer as the screen. Scrolling really takes some time and simply the wrong solution to this problem.

Related to all versions up to 1.8.9 on Windows and afaik also on Linux.

per1234, 2019-05-07:
@datenheim let me make sure I understand a couple of things:

When you say "board-manager", do you mean the contents of the **Tools > Board** menu? Boards Manager is something different (**Tools > Board > Boards Manager...**) so your terminology is not clear.

When you say "library manager", do you mean the contents of the **Sketch > Include Library** menu? Library Manager is something different (**Sketch > Include Library > Manage Libraries...** or **Tools > Manage Libraries...**).

>Scrolling really takes some time and simply the wrong solution to this problem.

Although not what you're requesting, this issue will be somewhat alleviated via https://github.com/arduino/Arduino/pull/8607

datenheim, 2019-05-09:
Exactly, the **Tools>Board** menu is meant, sorry for being not exact in the first place.
Same is true for **Sketch>Include Library**.
Yes, the feature mentioned in #8607 presenting recently used boards (libraries) more prominently will fit nicely with my request. There could well be a first submenu showing the recently used objects.

matthijskooijman, 2019-09-19:
For libraries, I implemented something similar in #1986, which divides puts libraries into submenus based on the current heading distinctions, as well as allowing them to be organized into subfolders under "libraries" (and these subfolders also cause submenus). The pull request was recently closed though, and was already quite outdated, but might still serve as inspiration in the future.

per1234, 2021-04-04:
The **Tools > Board** component of this request was fixed by https://github.com/arduino/Arduino/pull/9238 (thanks Matthijs!). The **Sketch > Include Library** component remains.