arduino/Arduino #8860: Reference content bundled with Arduino IDE is very outdated

per1234, 2019-05-09:
The Arduino IDE comes with a copy of the Arduino Language Reference and some library reference pages. These are used to provide documentation accessed via **Help > Reference** and the context menu's "Find in Reference".

The Arduino Language Reference content is at least 1.5 years out of date.

The library reference pages appear to be ~2 years out of date at least, since this change made Jul 17, 2017 is not present: https://github.com/arduino/Arduino/issues/6507
I didn't check back farther than that.

Since that time, many updates, corrections, and improvements have been made to the reference content. These files should be updated on every release of the Arduino IDE.

facchinm, 2019-05-14:
@per1234 the script we used to launch was  https://github.com/arduino/Arduino/blob/master/build/create_reference.pl . I think it is very outdated now and should be ported to the new reference architecture.
@cmaglie @mastrolinux @gvarisco any idea?

gvarisco, 2020-01-29:
Tagging @arduinosocial here

alranel, 2021-02-08:
I think we should just remove the offline version for now, and launch the online one when the reference menu item is selected.

PR #11495 (2021-08-18): Removed offline-reference docs.
The offline documentation is hard to update and gets outdated very quickly.

The `DocLinkGenerator` class was using the old `Base.showReference` method, but I found no way to actually trigger the callback. There may be some obscure way to call it, but it doesn't look correct to reach the Arduino website anyway, so I opted to remove it.

Fix #8860
commit 89eba8cf (2021-08-18): Removed bundled offline docs

The offline documentation is hard to update and gets outdated very
quickly.

Fix #8860
closed 2021-08-18 by cmaglie