arduino/Arduino #8861: Compile issue with whitespaces in path

fcooper, 2019-05-09:
Hi,

I'm running into an issue with multiple whitespaces in the ide's path. The problem appears to be with the code that is processing <ide path>/hardware/platform.txt. Something is causing it to strip multiple sequential white spaces and turn it into a single white space. When compiling any program I get the below error:

exec: "C:\\Users\\fcooper\\Downloads\\arduino -1.8.9-windows\\arduino-1.8.9\\tools-builder\\ctags\\5.8-arduino11/ctags": file does not exist

The full compile log can be found here: https://pastebin.com/nGx3vxAf

As you can see the ide is located at "C:\Users\fcooper\Downloads\arduino  -1.8.9-windows\arduino-1.8.9" not "C:\Users\fcooper\Downloads\arduino -1.8.9-windows\arduino-1.8.9". In other places the proper path in the log is seen except when calling ctags.

IDE Version: 1.8.9
OS: Windows 10 64-bit
Board: Arduino Uno WiFi Rev2
Board Package Version: 1.6.26
I'm running Arduino 1.8.9 on Windows 10.

facchinm, 2019-05-13:
Hi @fcooper ,
I've been able to reproduce it in windows; it looks like the issue is originated by this golang bug (https://github.com/golang/go/issues/17149) which is still under active investigation.
@per1234 can we move the issue to https://github.com/arduino/arduino-builder ?

per1234, 2019-05-13:
Moved to https://github.com/arduino/arduino-builder/issues/320

closed 2019-05-13 by per1234