arduino/Arduino #8863: [Board Manager] board packages previously installed will be deleted after new board packages installed.

konosubakonoakua, 2019-05-10:
I'm using Arduino IDE 1.8.9 and have ESP32 and STM32duino packages installed.
After I install the esp8266 package, `package_esp32_index.json` and `package_stm_index.json` are gone.
I have met this when I install esp32 after stm32duino got installed.
[image]
[image]

I copy the backup file into this folder and old packages are back now.
[image]
[image]
[image]
[image]

could you fix this please?

per1234, 2019-05-10:
Please make sure that you have all three package index URLs in your **File > Preferences > Additional Boards Manager URLs**, separated by commas. If you deleted the previous URLs from that preference when you added the ESP8266 URL, it would result in the exact behavior you've experienced.

Please let me know whether that was the problem.

konosubakonoakua, 2019-05-11:
Thank you, you are right, I try it, it works. thanks😁

closed 2019-05-11 by konosubakonoakua