arduino/Arduino #8864: Librairies  problem with update 1.8.9

simoue456, 2019-05-10:
I have this code working before the software update: 
void modbus_const_BROADCAST_1(byte packet) {

  delay(100);
  modbus_construct(&RTU_Packets_1[packet], 0, PRESET_MULTIPLE_REGISTERS, 117, 1, 1917, 1);
  delay(100);

and now I have this error :
EL-MC-Gen-001-CM-1.012018-09-25:2542:69: error: 'struct Packet' has no member named 'failedConnectionTime'

         boolTemp[0] = packetConnectionRefreshFunction(packetTemp_1->failedConnectionTime);

                                                                     ^

exit status 1
too many arguments to function 'void modbus_construct(Packet*, unsigned char, unsigned char, unsigned int, unsigned int, unsigned int)'

per1234, 2019-05-10:
Please provide a [minimal, complete, and verifiable example (MCVE)](https://stackoverflow.com/help/mcve) for the issue. Without that, it will be impossible for us to investigate your problem.

Make sure to provide all the information we would need to reproduce the error ourselves.

simoue456, 2019-05-10:
////////////////////      Librairies     ///////////////////

//Librairies de communication avec les capteurs
#include <SimpleModbusMaster.h>
#include <SimpleModbusMaster_2.h>

//Librairie pour communication I2C
#include <Wire.h>

////////////////////      Variables à configurer     ///////////////////

// Définition du nombre de données spéciales total     //-Config Section-//
// Par exemple, pour ELOPSYS courants anodiques,
// il y a 40 anodes sur une cuve, donc :
// #define nbDonneesSpecialsTotal 40
#define nbDonneesSpecialsTotal 18

//Définition du nombre de données spéciales maximum possible
//Cette variable est utilisé pour déclarer les variables relatifs aux
//anodes. Le maximum possible empeche l'overlapping des variables relatifs
//aux anodes.
#define nbDonneesSpecialsMax 60

// Données d'output max
// On défini ici le nombre maximum
// de sorties possible.
#define donneesOutputMax 64

//Points de mesures maximum par capteurs.
#define pointsMax 8

////////////////////      Envoi de configuration aux capteurs     ///////////////////

// Configuration 01 envoyée aux capteurs
// Par exemple :
// Résistivité, coefficient, distancPoints, etc.
int Config01[pointsMax];

// Configuration 02 envoyée aux capteurs
// Par exemple :
// Distance des points
int Config02[pointsMax];

// Configuration 03 envoyée aux capteurs
// Par exemple :
// Formule de 4ieme degré
int Config03[pointsMax];

// Configuration 04 envoyée aux capteurs
// Par exemple :
// Formule de 4ieme degré
int Config04[pointsMax * 2];

// Configuration capteurs
// Utilisé pour la création des packet
// Modbus RTU.
int ConfigCapteur[16];

// Données d'output
// Utilisé pour le transfert
// des outputs aux capteurs.
// Par exemple:
// Un capteur à 7 sorties digitales.
int donneesOutput[donneesOutputMax];

//Configurations du mode circuit ouvert
//Défini plus tard par le processeur maitre.
int ModeCircuitOuvertConstantH;
int ModeCircuitOuvertConstantC;
int ModeCircuitOuvertConstantMax;
[... 2912 lines cut ...]

EL-MC-Gen-001-CM-1.012018-09-25:2476:69: error: 'struct Packet' has no member named 'failedConnectionTime'

         boolTemp[3] = packetConnectionRefreshFunction(packetTemp_1->failedConnectionTime);

                                                                     ^

EL-MC-Gen-001-CM-1.012018-09-25:2515:69: error: 'struct Packet' has no member named 'failedConnectionTime'

         boolTemp[0] = packetConnectionRefreshFunction(packetTemp_1->failedConnectionTime);

                                                                     ^

EL-MC-Gen-001-CM-1.012018-09-25:2542:69: error: 'struct Packet' has no member named 'failedConnectionTime'

         boolTemp[0] = packetConnectionRefreshFunction(packetTemp_1->failedConnectionTime);

                                                                     ^

exit status 1
too many arguments to function 'void modbus_construct(Packet*, unsigned char, unsigned char, unsigned int, unsigned int, unsigned int)'

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

per1234, 2019-05-10:
Please provide a link to where you downloaded both of the SimpleModbus libraries. Or if you installed them via the Arduino Library Manager (**Sketch > Include Library > Manage Libraries...**) then say so and state the full name of the library.

Which version of the Arduino IDE were you using before updating to Arduino IDE 1.8.9?

simoue456, 2019-05-10:
before I use 1.6.5

per1234, 2019-05-10:
Please answer my other question.

simoue456, 2019-05-10:
https://github.com/angeloc/simplemodbusng

per1234, 2019-05-10:
That repository does not contain a file named SimpleModbusMaster_2.h but you include such a file in your sketch. Please tell me where you got the library that contains SimpleModbusMaster_2.h.

simoue456, 2019-05-10:
I can't find the link but I have this file
[SimpleModbusMaster_2.zip](https://github.com/arduino/Arduino/files/3167616/SimpleModbusMaster_2.zip)

per1234, 2019-05-10:
The same errors occur when compiling that code with Arduino IDE 1.6.5.

And that code is so far from the "minimal" demonstration I requested as to be ridiculous. 

If you need further assistance with this problem, please post on the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar). The forum has a limit of 9000 characters per post, so if you are going to stick with this massive sketch then you'll need to add it to the post as an attachment, but be aware that will make it less likely that you'll get help, since many forum members won't or can't download attachments.

closed 2019-05-10 by per1234

simoue456, 2019-05-10:
ok thanks for your help