arduino/Arduino #8866: COM ports not showing if connected to network

lemmingDev, 2019-05-10:
If I'm not connected to a network when opening the Arduino IDE, all COM ports are displayed under Tools / Ports correctly.

If I'm connected to a network, the ports menu is greyed out.
This behaviour occurs when I'm at work (a public highschool in Australia) which has pretty strict firewall and proxy settings that block a lot of things.

It happens on school desktops, along with my work laptop and student's personal laptops.

There must be something in the Arduino IDE that tries to call home on startup that my work is blocking that is causing this behaviour.

We shouldn't have to disconnect from the network to have the IDE work correctly. We need to access resources, mark rolls etc. I also can't have students ripping LAN cables out the back of desktop PCs every time the want to use the IDE.

Please allow COM ports to be listed even though computer has limited network connectivity.

facchinm, 2019-05-13:
Hi @lemmingDev ,
the discovery of serial ports should be totally unrelated with network ports. Can you provide a screencast of what is going on? Thanks!

PaulStoffregen, 2019-05-13:
Maybe related to getBoardWithMatchingVidPidFromCloud() ??

facchinm, 2019-05-13:
Unless the proxy/firewall leaves the connection pending that function should always return, in a way or another.

lemmingDev, 2019-05-14:
@facchimm

Screencast? Do you mean a video of what's going on?

Thanks for the reply

per1234, 2019-05-15:
>Do you mean a video of what's going on?

@lemmingDev that's correct. I'm using a free, open source, lightweight program called [LICEcap](https://www.cockos.com/licecap/) for this purpose.

lemmingDev, 2019-05-20:
Sorry - moving house and haven't had time to screencast. Not much to see anyway. When connected to network at school, Ports menu option is greyed out when opening Arduino as no ports are visible. When I disconnect from network and then reopen Arduino, ports menu is polulating the ports fine. Problem only happens at school.

Technicians at school say its probably a firewall issue.

What URLs does the IDE try to reach when opening?

facchinm, 2019-05-21:
It tries to reach:
http://api-builder.arduino.cc/builder/v1/boards/0xAAAA/0xBBBB (with AAAA and BBBB replaced by the USB VID and PID your board exposes).

That URL in a normal (non firewalled) environment should return 404 on a "board not found" and a json in case the board is known.

If the connection hangs in your environment it could be the root cause of the greyed out board list. 
Please note that the REST endpoint is called only if the board is not recognized (like ESP8266, which all share the same USB-to-serial converter)

lemmingDev, 2019-05-22:
Thanks. Will ask technicians to allow that URL.

The problem also presents itself when no board is attached though. Does it still try to reach that URL passing the currently selected board?

lemmingDev, 2019-05-31:
Wireshark capture indicates 52.139.250.253 getting blocked.

Looks like it belongs to Microsoft Azure

Waiting on regional technicians to continue looking into it.

Would be great if it was fixed on the Arduino end though so others don't have the problem

endorama, 2019-05-31:
Hello @lemmingDev, I'm part of the operations team at Arduino.  

The IP address you indicate is a server that does not belong to us. 

In order to help us debugging this issue would be possible for you to provide a `pcap` of a session where this problem arise with the Arduino IDE?  
As you may imagine is difficult to reproduce the network setup causing the issue, so a detailed Wireshark session could help us.

If possible stop any other network activity during the capture. Send the file to support@arduino.cc referencing "arduino/Arduino issue 8866".

Thank you!

lemmingDev, 2019-06-03:
@endorama 

Thanks for looking into this

Email with capture sent

endorama, 2019-06-04:
After further investigation, this seems mainly a fault not directly related to the IDE.

The IP address aforementioned is a IP address related to Windows Push Notification Service. Our guess is it gets blocked by network limitation on Microsoft telemetry.  The IDE was probably downloaded through Microsoft Store (I'm still waiting confirmation on this). This is a network issue on the client side we are not able to "fix". (We may be able to reproduce the issue but I'll continue the discovery and troubleshooting via private channel with @lemmingDev)

Could we try improving the Arduino IDE side of this issue, by ensuring that all network requests are terminated after a timeout? @facchinm would this be feasible?  
 
My understanding of this confirms what @facchinm said in the previous comment, this is likely a network issue and not a bug. I see a possible bug in how we handle "never ending" network requests, but will require further investigation.

endorama, 2019-06-25:
I can confirm the issue here was the local network topology preventing access to the required endpoint http://api-builder.arduino.cc/, resulting in pending connections.

I would close this as has been resolved, but I'd like to open a follow up improvement ticket to handle this errors more nicely.
@facchinm how effort would you think is required ? Is feasible with the current IDE network stack?

closed 2019-06-25 by endorama

facchinm, 2019-06-25:
Threading the request and canceling the thread on timeout should work, I'll try to setup a PR in the next few days

lemmingDev, 2019-06-30:
Fantastic!

Let me know when it's ready to test

lemmingDev, 2019-07-15:
Hi - Just spoke to regional IT support for my ork.

They say it may be a good idea for Arduino developers to make call to http://api-builder.arduino.cc/ and any other URLs proxy aware. They think this would allow the connection/response and fix the problem.

lemmingDev, 2019-08-09:
FYI - for anyone who may be experiencing the same issues, I added an entry in my hosts file to redirect 
api-builder.arduino.cc/
to 127.0.0.1 and it makes the ports get listed fine. 
Not the neatest solution, but will work while the developers hopefully work on making those calls proxy aware.