arduino/Arduino #8868: ESP8266 repository not working package_esp8266com_index.json

furbanow, 2019-05-11:
Hi,
I observed that from today i can not compile for my ESP8266 board.
It seems that the json for additionnal boards is malformed.
http://arduino.esp8266.com/stable/package_esp8266com_index.json

See how starts the json file. I guess it is a bug:
Because of this i can not work on my project.

_Dropping version 1.6.5-947-g39819f0
Dropping version 2.5.0-beta1
Dropping version 2.5.0-beta2
Dropping version 2.5.0-beta3
Dropping version 2.4.0-rc1
Dropping version 2.4.0-rc2
{
  "packages": [
    {
      "maintainer": "ESP8266 Community", 
      "help": {
        "online": "https://esp8266.com/arduino"
      }, 
      "websiteURL": "https://github.com/esp8266/Arduino", 
      "platforms": [
        {_
...

Thanks you

per1234, 2019-05-11:
ESP8266 core for Arduino is a 3rd party project. This problem is nothing Arduino can fix, nor is it our responsibility.

The problem has already been reported to the ESP8266 core for Arduino maintainers and they are working on fixing it:
https://github.com/esp8266/Arduino/issues/6068

closed 2019-05-11 by per1234

furbanow, 2019-05-11:
> esp8266/Arduino#6068 <https://github.com/esp8266/Arduino/issues/6068>

Thanks for that.