arduino/Arduino #8872: How to update my library

junwha0511, 2019-05-12:
6 months ago, I issued to enroll my library into arduino library manager.
and now I want to update my library 1.0.0 to 1.1.0.

how can I do it?

just update my github? or new Issue?

junwha0511, 2019-05-12:
if second, please update my library into Arduino Library Manager.

per1234, 2019-05-12:
Here is the process:

1. Update the `version` value in library.properties to 1.1.0.
2. Create a GitHub release or Git tag.

The Arduino Library Manager indexer job regularly scans all the repositories in the Library Manager index for new tags and automatically picks up valid tags and adds them to the index. So there is no need to submit an issue here for updates. It's only necessary to submit the initial issue to get your repository added to the index and everything is automatic after that.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list

closed 2019-05-12 by per1234