arduino/Arduino #8881: Arduino WiFi Rev2 not detected on iMac running Win 10

PickyBiker, 2019-05-14:
Although I have successfully used several types of Arduino and nodeMCU ESP8266 on an iMac running Windows 10, an Arduino UNO WiFi Rev2 is not detected as a COM port even after installing the Mega AVR package and it's associated windows driver.
Please see the Arduino forum issue regarding this issue and the details. 
[https://forum.arduino.cc/index.php?topic=615506.0](url)

facchinm, 2019-05-15:
The drivers in the installation package are just needed for uploading (since upload is based on HID), on Windows10 the serial port should be recognized without any driver. 

My guess are either a faulty USB cable or a faulty board. To make sure it's not a driver problem, can you test if it gets recognized in OSX (since you have it installed anyway)?

Then, try to use another USB cable and please report your findings. Thanks you

PickyBiker, 2019-05-15:
If I boot in Mac OS, the board is detected and works just fine. 
If I install the board and cable on a standard PC with win 10, the board works fine.
When I boot the iMac into Windows 10, (without the driver) it is detected as "USB Serial Device" under Ports but no Com Port is assigned.
If I reinstall the windows driver, the board is detected as "mEDBG Virtual Com Port", but no Com port (Com1, Com2 etc.) is assigned. 

The board and the cable are the same in all tests. 

Please let me know if you need any further information.

PickyBiker, 2019-05-29:
Based on feedback from facchinm, I did a complete wipe of my mac hard drive, reinstalled Mojave and then reinstalled Win 10 on Bootcamp. The system now detects and assign COMn ports to the boards that I formerly reported as not working. 

I have closed this issue.

closed 2019-05-29 by PickyBiker