arduino/Arduino #8882: [Library Manager] Please add Virtuino libraries to Library Manager

iliaslamprou, 2019-05-14:
I want to add the following libraries to arduino library manager
https://github.com/iliaslamprou/virtuino
https://github.com/iliaslamprou/virtuinoESP
https://github.com/iliaslamprou/virtuino_stm32

per1234, 2019-05-15:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

iliaslamprou, 2019-05-15:
Thank you!

I have finished the necessary changes.
I think it is ok now
Could you add Virtuino libraries to the Library Manager?

*Best Regards*
*Ilias Lamprou*
*Virtuino Developer*

*https://virtuino.com <http://virtuino.com>*
*Play Store page*
<https://play.google.com/store/apps/developer?id=Ilias+Lamprou>

closed 2019-05-20 by iliaslamprou