arduino/Arduino #8894: Add autocomplete from this fork

anselal, 2019-05-18:
Could you add an autocomplete feature to your IDE please?

There is a fork at https://github.com/OpenDevice/Arduino which exists several years ago which has implemented this feature.

Could you perhaps merge his codebase and give this guy a credit?

per1234, 2019-05-18:
Autocomplete functionality has already been added in the Arduino IDE Beta Build:
https://blog.arduino.cc/2017/10/11/be-among-the-first-to-try-arduino-ide-1-9-beta/
>- Initial support for autocompletion (activate it using CTRL+space)
>    - Attention: Launching for the first time is quite slow and will freeze the UI. Don’t worry, simply wait for it to unstick.

You can download the Arduino IDE Beta Build here:
https://www.arduino.cc/en/Main/Software#beta

anselal, 2019-05-18:
thnx.

closed 2019-05-21 by per1234