arduino/Arduino #8898: Please add library for Witty 2WD balancing bot

boxtec, 2019-05-20:
Please add our support library for the Witty 2WD bot to the library manager:

https://git.boxtec.ch/didel/Witty

Thank you in advance.

per1234, 2019-05-20:
Hi. We only support addition of libraries from major git-hosting websites like GitHub, BitBucket or GitLab. If you would like to move your website to GitHub, we could proceed with adding it to the Arduino Library Manager index. Let us know once this is done and the new URL.

If you are not willing to move the repository, you can go ahead and close this issue report.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager

boxtec, 2019-05-21:
Thank you for your swift follow-up and explanation. 
New repo URL is: https://github.com/boxtec/Witty

Thanks.

cmaglie, 2019-05-25:
done

closed 2019-05-25 by cmaglie