arduino/Arduino #8899: Arduino DUE problem with DAC0 and DAC1

laboIsep, 2019-05-20:
Read the Issues section of the Contributing Rules at the "contributing guidelines" link above before submitting an issue report.

laboIsep, 2019-05-20:
We use an DUE arduino card and when compiling " analogWrite(DAC0, 100) "

there is an error:
 👍 Arduino : 1.8.9 (Windows 10), Carte : "Arduino Duemilanove or Diecimila, ATmega328P"
Test_LMS_Opt:41:16: error: 'DAC0' was not declared in this scope
    analogWrite(DAC0,100);
exit status 1
'DAC0' was not declared in this scope

thanks for your answer

Hervé

per1234, 2019-05-20:
Due != Duemilanove. These are completely different boards, thus your error.

If you need more assistance with this issue, please use the Arduino Forum, which is a more appropriate place for support requests. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.

closed 2019-05-20 by per1234