arduino/Arduino #8904: LCD Hello World tutorial: missing connection under "Circuit" heading.

Vic1970, 2019-05-21:
In the LCD Hello World tutorial at [https://www.arduino.cc/en/Tutorial/HelloWorld](https://www.arduino.cc/en/Tutorial/HelloWorld) , 

under the section "Circuit", the connection of LCD RW to ground is **omitted**.

This connection is shown in the schematic and the Fritzing diagram, and even listed in the code, but not in the written list of connections to make.

Additionally, the connection of the 10k pot could be reformatted to follow the same bulleted list format for connections as the rest of the written circuit list.

per1234, 2019-05-21:
Thanks so much for your suggestions for improvements to the documentation @Vic1970!

This was also reported way back in 2010 and closed as fixed:
https://github.com/arduino/Arduino/issues/402
I notice that, although the title of that issue refers to the pin as R/W, in the text of the issue they call it "LCD RS pin", so maybe that was the cause of the issue being closed as fixed even though the R/W pin problem was not?

Vic1970, 2019-05-21:
Hi, yes, there are both RS and RW pins on the LCD pinout so that's probably the source of the confusion.  
Thank you for adding the label, I am new at this.

closed 2020-09-29 by dcuartielles

nitheesh2008, 2020-10-05:
hi