arduino/Arduino #8905: [Library manager] please add simple PWM set frequency

guru-florida, 2019-05-22:
I wrote a library that has the setPwmFrequency() function...possibly others to come. Please add to library manager. Thanks!

https://github.com/guru-florida/PWMFreak

per1234, 2019-05-22:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

guru-florida, 2019-05-23:
DONE.   
(Doh. Not my first time and I forgot. lol)

cmaglie, 2019-05-25:
done

closed 2019-05-25 by cmaglie