arduino/Arduino #8907: Arduino IDE boards download erro

XingyumLee, 2019-05-23:
this is what i got below,can anyone know how to fix it.thanks

 Error downloading https://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading https://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:333)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:307)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:285)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:146)
	... 1 more
Caused by: java.nio.file.FileSystemException: C:\Users\snowy\AppData\Local\Arduino15\package_index.json.tmp: 另一个程序正在使用此文件，进程无法访问。

	at sun.nio.fs.WindowsException.translateToIOException(WindowsException.java:86)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:97)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:102)
	at sun.nio.fs.WindowsFileSystemProvider.implDelete(WindowsFileSystemProvider.java:269)
	at sun.nio.fs.AbstractFileSystemProvider.delete(AbstractFileSystemProvider.java:103)
	at java.nio.file.Files.delete(Files.java:1126)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:156)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more

per1234, 2019-05-23:
This sort of error could be caused by your antivirus software. Try **TEMPORARILY** disabling your antivirus while running Boards Manager to see if the problem goes away, then turn the antivirus back on immediately afterwards. If the problem doesn't occur with the antivirus turned off, you will need to adjust the settings of your antivirus to whitelist the appropriate file, folder, or process so it doesn't interfere with Boards Manager.

per1234, 2019-09-16:
@XingyumLee please let us know if the issue is resolved so we can close this.

per1234, 2020-08-04:
Closing due to lack of response.

@XingyumLee if you're still having the issue and will provide the requested information I'll be happy to re-open this issue.

closed 2020-08-04 by per1234