arduino/Arduino #8909: Arduino Wifi Rev2 Manufacturing error Component C904

PickyBiker, 2019-05-23:
There is a manufacturing problem with the Arduino UNO WiFi Rev2 board involving a wrong component. C904 is a 20pf capacitor on the crystal for the RTC. On at least 3 boards from two or three sources, there was a 19.1k resister where that capacitor should be. One was a board I bought from Amazon that I returned.  A second one was bought directly off the Arduino online store. Both have that problem. There is a third one reported in some detail and with pictures in this library issue: [TimeLib.h updated for use RTC circuit embedded with megaavr #27](url). 
I am posting here in the hope that someone can confirm the right people at Arduino are aware of  this and confirm that changes to the manufacturing process is being taken.

PickyBiker, 2019-05-24:
Good news. Arduino customer support has confirmed they know about this problem and their developers are working to correct it.

agdl, 2019-09-23:
@PickyBiker we fixed that batch. Thanks for reporting this

PickyBiker, 2019-09-23:
agdl, I bought one last week from Arduino USA and it was bad. After 3 orders, all bad, Arduino Customer Support arranged to have a verified fixed one shipped to me directly from Italy.  So the problem is not yet corrected in all distribution centers.