arduino/Arduino #8912: IDE 1.6.5

simoue456, 2019-05-24:
I I have the IDE 1.8.8 but I need IDE 1.6.5 version. When I install this version of compatibility issues occur on my computer (windows 7) know how to adjust this?

gvarisco, 2019-05-24:
Hi @simoue456,

could you please provide us with a screenshot (or log) of such 'compatibility issues' in order to let us understand what is going on?

1.6.5 can be downloaded at https://www.arduino.cc/download_handler.php?f=/arduino-1.6.5-windows.exe but we'd definitely be interested in letting you use 1.8.x as it contains lot of bug fixes and improvements.

simoue456, 2019-05-27:
I can't ake a screenshot because whent I double clic on the program he never open

Pimmetje, 2019-05-28:
If the whole IDE does not start it's more likely a arduino issue. In that case u could try to rename the arduino folder in your My Documents folder to arduino2 (a new one will be created when u start the 1.6.5 IDE).

If some code gives issue's please provide a sample code.

per1234, 2019-09-16:
Closing due to lack of feedback.

closed 2019-09-16 by per1234