arduino/Arduino #8918: [Library Manager] Please add libraries

obdevel, 2019-05-26:
Please add the following:

https://github.com/obdevel/CBUS
https://github.com/obdevel/CBUS2515
https://github.com/obdevel/CBUSconfig
https://github.com/obdevel/CBUSLED
https://github.com/obdevel/CBUSswitch

Thank you

per1234, 2019-05-27:
In each of those repositories, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

obdevel, 2019-05-27:
I've now corrected this and created a 1.0.0 release for each library. Thank you.

cmaglie, 2019-06-07:
done

closed 2019-06-07 by cmaglie