arduino/Arduino #8920: [Library Manager] Please Add Library

jaenrig-ifx, 2019-05-27:
Please add following library to library manager: 

https://github.com/Infineon/TLx4966-Direction-Speed-Sensor

jaenrig-ifx, 2019-05-27:
Please also these other libraries:
https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor
https://github.com/Infineon/TLE5012-Magnetic-Angle-Sensor
https://github.com/Infineon/TLI4970-D050T4-Current-Sensor
https://github.com/Infineon/RGB-LED-Lighting-Shield-XMC1202
https://github.com/Infineon/Stepper-Motor-Shield-IFX9201-XMC1300
https://github.com/Infineon/OPTIGA-Trust-E-Security-Controller

Thanks!!

cmaglie, 2019-06-07:
done

closed 2019-06-07 by cmaglie

jaenrig-ifx, 2019-06-07:
Thanks a lot!!

njh, 2022-03-01:
@jaenrig-ifx it looks like this repo has been deleted:
https://github.com/Infineon/OPTIGA-Trust-E-Security-Controller

Do you have any information about why?

jaenrig-ifx, 2022-03-01:
Hi @njh,

Maybe @ayushev have some information about it?