arduino/Arduino #8924: when update win 10 to 1901 version arduino IDE dont work normal and dont detect device

elagh, 2019-05-28:
i update win 10 to 1901 version, Then the software knows the common port, but does not identify the arduino device ( arduino pro mini 5 v). 
(i test the unio and mega series but The problem remained.)
and Arduino IDE show the ""avrdude: ser_open(): can't open device "\\.\COM10": Access is denied."" error.
first i think my device has problem but change the devices but The problem remained until I reviewed the program from another computer with the Windows version 1809 and there was no problem.
* before update windows  i dont have any problem.

facchinm, 2019-05-29:
Hi @elagh ,
it Win10 1901 a real version? I could only find 1903 as the latest available and I'm preparing a PC with that version to test the issue.

elagh, 2019-05-29:
hi. am sorry that version is 1903.
I typed mistakes.
thank.

elagh, 2019-06-10:
hi.
new error in arduino

avrdude: ser_open(): can't open device "\\.\COM4": Access is denied.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
processing.app.SerialException: Error opening serial port 'COM4'.
	at processing.app.Serial.<init>(Serial.java:147)
	at processing.app.Serial.<init>(Serial.java:82)
	at processing.app.SerialMonitor$4.<init>(SerialMonitor.java:101)
	at processing.app.SerialMonitor.open(SerialMonitor.java:101)
	at processing.app.AbstractMonitor.resume(AbstractMonitor.java:104)
	at processing.app.Editor.resumeOrCloseSerialMonitor(Editor.java:2126)
	at processing.app.Editor.access$1300(Editor.java:115)
	at processing.app.Editor$UploadHandler.run(Editor.java:2104)
	at java.lang.Thread.run(Thread.java:748)
Caused by: jssc.SerialPortException: Port name - COM4; Method name - openPort(); Exception type - Port busy.
	at jssc.SerialPort.openPort(SerialPort.java:164)
	at processing.app.Serial.<init>(Serial.java:136)
	... 8 more
Error opening serial port 'COM4'.

dahu74fr, 2019-07-04:
Hi,
Have you found an issue?

dahu74fr, 2019-07-04:
hi,
Thanks for your reply.
So i think we can wait for a long time!!
For me, i can work with a chinese clone with CH340.
It works fine and no problems of driver.

elagh, 2019-07-04:
hi.
your welcome
be good luck.

feikname, 2019-07-04:
`Caused by: jssc.SerialPortException: Port name - COM4; Method name - openPort(); Exception type - Port busy.`

Only _one_ thread can access a COM port at a time. Quite likely another program is using your COM port, please download [the Portmon utility](https://docs.microsoft.com/en-us/sysinternals/downloads/portmon) from Microsoft and check if you can see any process as the culprit for using your COM port.

elagh, 2019-07-04:
hi.
thank for your reply.
i download and check this but not device using COM port.