arduino/Arduino #8931: keywords.txt highlight

comertial, 2019-05-31:
keywords.txt files in libraries should highlight only if the library they exist is included in the sketch.. It's frustrating to see the highlights of every library you have installed, even when you are not using it. Especially if you have installed so many libraries!

PaulStoffregen, 2019-06-12:
There are 2 pieces to this feature.  On the IDE side, I can help with changing the highlighting.  I sent the pull request which changes the keywords when you switch boards/platforms, so I'm pretty familiar with what to do on the Java side.

The other side is actually parsing the #include lines.  I believe the non-beta IDE is still using ctags, but the plan is to switch to clang or llvm parsing.  Either way, it's external from the IDE.  So a big question is whether the IDE will detect the change when you've added another #include line in the editor, or it the recognition of the new include will happen only when you click Verify or Upload, or perhaps if the IDE will run the external parsing at other times?  I'm not very familiar with Google Go language, nor have I been involved much in all that stuff since it broke off from the IDE around 1.6.65 to 1.6.6.

facchinm, 2019-06-12:
Probably the biggest problem is that we should execute a "dry run" compilation (just to discover libraries) every time a new include is added.
So it could be split into two parts:
* a listener on java side firing an event on "new header added/removed"
* a commandline switch for the builder/cli that just resolves the libraries in the background and returns a (json)?
* a function that reloads keyworks.txt by only considering the ones provided by included libraries.

The list of correct keywords could also be directly provided by the builder. 
@PaulStoffregen is you are willing to implement the Java part I can take care of the golang one.