arduino/Arduino #8934: Documentation of SPI begin() incorrect

technoblogy, 2019-06-01:
The documentation of SPI **begin()** says:

> Initializes the SPI bus by setting SCK, MOSI, and SS to outputs, pulling SCK and MOSI low, and SS high.

I think **begin()** doesn't set SS to an output and make it high; you have to do this yourself. In fact **begin()** doesn't even know which the SS pin is.

per1234, 2019-06-01:
>I think begin() doesn't set SS to an output and make it high

I'll take Arduino AVR Boards' SPI library as an example, since it's the most commonly used.

Here you can see the code that sets `SS` to `HIGH`:
https://github.com/arduino/ArduinoCore-avr/blob/1.6.23/libraries/SPI/src/SPI.cpp#L36-L40
```c++
    // if the SS pin is not already configured as an output
    // then set it high (to enable the internal pull-up resistor)
    if(!(*reg & bit)){
      digitalWrite(SS, HIGH);
    }
```
Here you can see the code that sets `SS` to `OUTPUT` mode:
https://github.com/arduino/ArduinoCore-avr/blob/1.6.23/libraries/SPI/src/SPI.cpp#L42-L45
```c++
    // When the SS pin is set as OUTPUT, it can be used as
    // a general purpose output port (it doesn't influence
    // SPI operations).
    pinMode(SS, OUTPUT);
```

>In fact begin() doesn't even know which the SS pin is.

`SS` is defined in the variant file:
https://github.com/arduino/ArduinoCore-avr/blob/1.6.23/variants/standard/pins_arduino.h#L38
```c++
#define PIN_SPI_SS    (10)
```
https://github.com/arduino/ArduinoCore-avr/blob/1.6.23/variants/standard/pins_arduino.h#L43
```c++
static const uint8_t SS   = PIN_SPI_SS;
```
I think you're confusing the SS (slave select) pin of the Arduino board's microcontroller with the pin used to control the SS (AKA CS for "chip select") pin of the SPI slave device attached to that microcontroller. That's easy to do because the Arduino board's SS pin is often used to control the device's SS pin (though this is not required). We do document this on the SPI library's home page:
https://www.arduino.cc/en/Reference/SPI

closed 2019-06-01 by per1234

technoblogy, 2019-06-01:
Thanks for the explanation, and sorry for the incorrect issue report.