arduino/Arduino #8935: [Library Manager] Please add libraries ss_oled and BitBang_I2C

bitbank2, 2019-06-01:
https://github.com/bitbank2/ss_oled
https://github.com/bitbank2/BitBang_I2C

bitbank2, 2019-06-06:
Please add another Arduino library:
https://github.com/bitbank2/bb_hx1230

per1234, 2019-06-06:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file in your bb_hx1230 repository.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

bitbank2, 2019-06-06:
Added a release (thanks for the reminder)

cmaglie, 2019-06-07:
done

closed 2019-06-07 by cmaglie