arduino/Arduino #8936: Downloaded boards disappeared

Jivansha, 2019-06-01:
There is a problem with the functioning of the Arduino Ide relating to how boards are downloaded. I had perfectly working board libraries available on my Ide until today when all of my installed boards disappeared and they can't be reinstalled again.
A similar issue has been faced by many others(with regard to post) with no solution -
https://forum.arduino.cc/index.php?topic=528806.0
What is the possible way of rectifying it?

per1234, 2019-06-02:
If you open **Tools > Board > Boards Manager** and then wait for the downloads to finish, do you get any error messages?

Jivansha, 2019-06-02:
After a few seconds after starting the installation of the previously installed esp8266 board it says- 
"Download from http://arduino.esp8266.com/stable/package_esp8266com_index.json failed."
at the bottom of the board manager. I have been using Nodemcu before.

Jivansha, 2019-06-02:
Hi, this is to update that now the boards can be installed again. No changes had been made. Everything is as it was. But I would like the community to look into the issue so that it is not faced by users in future.

per1234, 2019-06-02:
I'm going to reopen this because I agree it's something that should be fixed and surprisingly I couldn't find an existing open issue about it.

The problem is that the Arduino IDE deletes each .json file before attempting the download. So if you run Boards Manager while a Boards Manager URL for an installed package is temporarily unavailable then the boards of that package disappear from the **Tools > Board** menu. The fix would be to only delete the existing .json file if the URL can be accessed. I see the IDE creates a .json.tmp file so the system is already in place for this to work correctly.

This is something that has been reported quite a few times on the forum, especially for a while there when the ESP8266 URL was having frequent outages.

matthijskooijman, 2019-06-03:
A simple fix would be to only remove an old json file when a new one was succesfully downloaded. However, I think that still means that when the maintainer of a board removes some version (or other data, I don't know the data off-hand) from the JSON, this could also break existing installed boards.

I wonder if, when installing a boards, the relevant bits from the JSON should be copied into an "installed.json" or something, so that even when all external json files become corrupted, empty, deleted, or whatever, the installed boards can still be used normally?

Jivansha, 2019-06-03:
The solution seems right but since the board is to be connected to wifi for most of its operations, it is possible this requires regular updating of the network state and then, in this case, the solution may again faulter?

facchinm, 2019-06-17:
@matthijskooijman I love the `installed.json` solution, since it saves both the situations. 
@cmaglie @masci is it something we can adopt in the cli as well?

masci, 2019-06-17:
@facchinm going to track this as a feature request for the CLI, the issue might indeed be the very same as the IDE.

facchinm, 2019-09-20:
Should be fixed by #9023

closed 2019-09-20 by facchinm