arduino/Arduino #8937: [Library Manager] LiquidCrystal_PCF8574 library has copyright & LGPL violations

bperrybap, 2019-06-01:
I'm not sure how Arduino handles copyright and licensing issues for libraries added to the library manager. IMO, the library should be removed from the library manager.

The library LiquidCrystal_PCF8574 in the library manager has copyright and LGPL license violations.
The author @mathertel (Matthias Hertel) started with code that was licensed as LGPL 2.1+ and made modifications to it creating a derivative but then has attempted to change the license terms on the derivative to a license that is not allowed under the LGPL 2.1+ licensing terms.
The author  is aware of this and has been for quite some time (since Nov 2017) and so far has been unwilling to correct this.
Here are the github issues where the author was made aware of the issues and how he reacted.
https://github.com/mathertel/LiquidCrystal_PCF8574/issues/3
https://github.com/mathertel/LiquidCrystal_PCF8574/issues/4

The author is wanting to re-license his derivative work with BSD licensing terms rather than LGPL by adding this text to the code:
```
/// From Author:
/// The extended work is licensed under a BSD style license.\n
/// See http://www.mathertel.de/License.aspx
/// 
/// Both License rules apply.
```
This change of licensing terms not allowed by the original copyright holder's LGPL licensing terms.

IMO, the Arduino team should not allow this type of flagrant disregard of licensing terms and should remove the library from the library manager until the author corrects the issues.
From his github page: https://github.com/mathertel
Here is an email contact address for @mathertel : mathertel@hotmail.com

mathertel, 2019-06-06:
Problems solved by re-implementing.

bperrybap, 2019-06-06:
You still have a minor issue on your copyright notices.
Your github library license file is 2005 to 2016 and your license web page referenced is claiming 2005 to 2012
Both are incorrect as the copyright now begins in 2019.

I would  recommend that you use the github recommend license file format. 
github uses the ruby licensee script to auto detect the license type. 
https://help.github.com/en/articles/licensing-a-repository (see the **Selecting a license** section)
https://github.com/licensee/licensee
If github can detect the license type,  your repository will the show license type directly on the main repository page instead of just providing a link to view the LICENSE file.

closed 2019-06-25 by per1234