arduino/Arduino #8938: Arduino IDE 1.8.9 crash on start. No output from arduino_debug.exe

Manusisko, 2019-06-02:
I had been using Arduino IDE 1.8.9 for some 2 weeks now, until last night when Windows Crashed and re-started. Since then, I've tried to start up Arduino IDE but it crashes after showing the splash screen with no error messages.
I've tried:
1. Rebooting.
2. Unninstalling and Re-installing Arduino IDE.
3. Running arduino_debug.exe from cmd but it outputs no message at all.

Anyone has any ideas?

closed 2019-06-02 by per1234

feikname, 2019-06-02:
@per1234 was this close accidental?

per1234, 2019-06-03:
@feikname no. It was intentional because @Manusisko said "This was a Windows Issue", which was affecting several other applications. It seems to me that there is nothing Arduino could do to fix the issue so there is no point in leaving the issue open.

If I'm wrong about that, I'd be happy to re-open.