arduino/Arduino #8939: Having some trouble updating library for arduino managed library

s-m-t, 2019-06-03:
Hello,
I'm having some trouble getting the arduino package manager to update my library: https://github.com/arbotics-llc/databot_arduino

Part of the problem may be that in the initial version the url field in library.properties I mistyped the url as /databot-arduino with a hyphen instead of /databot_arduino with an underscore. Could this confuse the auto updater? Another issue may be that in trying to get it to work I deleted and remade multiple releases... Thanks.

per1234, 2019-06-03:
The reason why your 0.1.1 release was not picked up by the Library Manager indexer is because you forgot to update the `version` field of library.properties:
https://github.com/arbotics-llc/databot_arduino/blob/v0.1.1/library.properties#L2

>Part of the problem may be that in the initial version the url field in library.properties I mistyped the url as /databot-arduino with a hyphen instead of /databot_arduino with an underscore. Could this confuse the auto updater?

I don't believe so.

Although the indexer job runs hourly, it may take a couple more hours for the updated index file to propagate to all servers. I don't see anything wrong with your 0.1.2 release. I think it will just take a little more time before it becomes available for installation via Library Manager.

s-m-t, 2019-06-03:
Ok cool, I'll check again in the morning. Thank you

s-m-t, 2019-06-03:
It updated, looks like it may have been because I had a newline in the paragraph section of library.properties?

per1234, 2019-06-04:
You're right! I didn't notice that.

closed 2019-06-04 by per1234