arduino/Arduino #8951: Place an explicit warning that a resistor is mandatory when conecting LEDs to output pins

gaazkam, 2019-06-06:
Related to https://github.com/arduino/Arduino/issues/8950 .

In my university I was told that resistors are not needed when LEDs are connected to output pins. According to my academic, while connecting a Led to a 5V or 3.3V power pin on one side and ground on the other side would burn things up, connecting a led directly to an output pin on one side and ground on the other side was safe and did not require resistors. Therefore, to complete homework, I was given an Arduino board, a set of LEDs and cables, but no resistors.

Then I started hearing from other people that my academic was wrong and that I was damaging my university's property - both the Arduino board and LEDs. (For example, I got linked to [this](http://www.thebox.myzen.co.uk/Tutorial/LEDs.html) ).

If this misconception is as rampant as it seems and as untrue as people say, let me propose adding an explicit warning to the docs that a resistor is actually mandatory and that not using it is likely to cause damage to the hardware; perhaps also even add an explicit notice like "regardless of whatever anyone may tell you", thus acknowledging the widespreadeness of the misconception. [The official Blink tutorial](https://www.arduino.cc/en/Tutorial/Blink) seems like a good place for such a notice.

animeshsrivastava24, 2019-06-08:
@per1234  Please guide is it possible for me to fix it.

per1234, 2019-06-08:
@animeshsrivastava24 thanks for your interest in helping with this! The way you can help with this is by providing a clear description of the changes you propose be made to https://www.arduino.cc/en/Tutorial/Blink in a reply here. 

@gaazkam is welcome to give feedback on your proposal.

This approach will make it faster for an Arduino employee with edit access to the tutorial content to make the desired change.

ElectricRCAircraftGuy, 2019-06-08:
@gaazkam, your professor is wrong: not using a resistor will likely destroy the pin or the LED, or both. The calculation is simple: V = IR (Voltage = Current x Resistance), or solving for I: I = V/R. Notice that as R approaches zero, I will approach infinity, and hence your current goes uncontrolled and blows something. 

However, you can use an internal 20k pull-up resistor that is built into the Arduino, and complete your assignment right now anyway, as follows:

```
void safeDigitalWrite(uint8_t pin, bool state)
{
    if (state == HIGH)
    {
        // Write pin HIGH through its internal ~20k resistor
        pinMode(pin, INPUT_PULLUP);
    }
    else
    {
        // Output LOW
        pinMode(pin, OUTPUT);
        digitalWrite(pin, LOW);
    }
}

// Now you can blink a pin with:

void setup()
{
    // no need to call pinMode()
}

void loop()
{
    safeDigitalWrite(LED_BUILTIN, HIGH);
    delay(1000);
    safeDigitalWrite(LED_BUILTIN, LOW);
    delay(1000);
}
```

I checked the source code for `pinMode()` in the latest version of Arduino to be sure, and the following is also legitimate and safe in place of the (better) `safeDigitalWrite()` function above:
```
// Note that when the state is LOW, the pin isn't actually connected to ground--it is just floating--but the LED will still turn off
#define safeDigitalWrite(pin, state) ((state) == HIGH ? pinMode(pin, INPUT_PULLUP) : pinMode(pin, INPUT))
```

Lastly, the following is also legitimate and works, and functions exactly like the `#define` version above:

```
void setup()
{
    pinMode(pin, INPUT);
}

void loop()
{
    digitalWrite(LED_BUILTIN, HIGH); // pin mode is INPUT, so this just connects the pin to 5V through the internal ~20k pullup resistor (same as `pinMode(pin, INPUT_PULLUP)`)
    delay(1000);
    digitalWrite(LED_BUILTIN, LOW); // turn internal pullup resistor OFF (pin is now floating, but the LED will turn off)
    delay(1000);
}
```

To @per1234, if you give me edit rights I will write a new tutorial called **Blink With Internal Resistor** (if it doesn't' already exist).

animeshsrivastava24, 2019-06-09:
@gaazkam @ElectricRCAircraftGuy 

> The internal pull-ups only effect the pin if it is configured as an input pin, it has no bearing on how an output pin behaves. An output pin would still require a series connected current limiting resistor to safely drive a LED. The resistor needs to be sized to limit the current to the maximum value allowed for the LED or 40ma to protect the output pin, which ever is less.

> The internal pull ups can save you from needing an external pull up resistor to interface with switch contacts. The pull-up will insure that there is a consitant logic HIGH when the switch is open and will read as a LOW when the switch contacts are closed if the other end of the switch is wired to ground. There is no way to change the value of the internal pull up resistors, they are what are, fixed and of high resistance.

Find more discussion at : https://forum.arduino.cc/index.php?topic=13589.0

@per1234 , sure I am working on the description and will post it soon:

> The way you can help with this is by providing a clear description of the changes you propose be made to https://www.arduino.cc/en/Tutorial/Blink in a reply here.