arduino/Arduino #8960: Urduino upload error report

rontapp, 2019-06-09:
Read the IArduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Arduino/Genuino Uno"

Build options changed, rebuilding all
C:\Users\Siliconium\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp: In member function 'drawFastHLine':

C:\Users\Siliconium\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp:1565:1: internal compiler error: Segmentation fault

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files/windowsapps/arduinollc.arduinoide_1.8.21.0_x86__mdqgnx93n4wtt/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
ssues section of the Contributing Rules at the "contributing guidelines" link above before submitting an issue report.

rontapp, 2019-06-09:
what have I done? all was well then this.
I deleted the app Arduino IDE fro my pc the re installed it can get rid of this error

per1234, 2019-06-09:
Closing as duplicate of https://github.com/arduino/Arduino/issues/7949

Here's a workaround:
1. Tools > Board > Boards Manager
1. Wait for downloads to finish.
1. When you move the mouse pointer over "Arduino AVR Boards", you will see a "Select version" dropdown menu appear. Select "1.6.21".
1. Click "Install".
1. Wait for installation to finish.
1. Click "Close".

If you have **File > Preferences > Check for updates on startup** checked, the Arduino IDE may occasionally notify you that a new version of Arduino AVR Boards is available, you'll need to refrain from updating back to the new Arduino AVR Boards version, otherwise you'll be back to seeing the segmentation fault error again.

closed 2019-06-09 by per1234

rontapp, 2019-06-09:
Thank you so much!