arduino/Arduino #8961: I would like to present a new library for Arduino containing PID controllers.

KorneliusThomas, 2019-06-09:
Read the Issues section of the Contributing Rules at the "contributing guidelines" link above before submitting an issue report.

KorneliusThomas, 2019-06-09:
I wait for comments.

KorneliusThomas, 2019-06-11:
There is an error in library URL.
The right library URL is: https://github.com/KorneliusThomas/PIDcontrollersModular
[PIDArduinoControllersModular.pdf](https://github.com/arduino/Arduino/files/3274508/PIDArduinoControllersModular.pdf)

per1234, 2019-06-11:
@KorneliusThomas did you want to request addition in the Arduino Library Manager index, or are you just advertising your library and asking for feedback?

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

KorneliusThomas, 2019-06-12:
I would like to add the library PIDcontrollersModular in the Arduino Library Manager.regardsKornelius

https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

per1234, 2019-06-12:
There are a couple things you need to do before your library can be added to the Arduino Library Manager index:

1. Move the library out of the .7z file and delete the .7z file. Beyond needing to do this for Library Manager, it's doesn't make sense to put a library on GitHub inside a compressed file. GitHub automatically generates .zip files of the repository, which are accessible via the **Clone or Download > Download ZIP** button on the home page as well as for each release or tag. Having the library in a compressed file defeats all the great version control and collaborative features of Git and GitHub.
2. Make sure the library is in the root of the repository, not in a subfolder. This means the library.properties file should be in the root folder of the repository.
3. Create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

KorneliusThomas, 2019-06-13:
Hi,I have modified my repositoryaccording to your suggestions.Please check it.Is it correct now?
regardsKornelius

per1234, 2019-06-13:
No, it's not correct. The library must be in the root of the repository. Your repository currently has the library under the PIDcontrollersModular, which is not allowed. Please move the library up to the root of the repository and then either create a new tag/release or else redo the existing tag/release.

Once that's done, please reply here so we'll know we can proceed.

KorneliusThomas, 2019-06-14:
Hi,I hope now is correct.regardsKornelius

KorneliusThomas, 2019-06-20:
Hi,
since some time I try to adapt my library to Arduino standards.
We have exchanged few emails, but now the contact is broken.
Last time I have done few modification in the repository and send a
email to Arduino, but got no reply. Please inform me about the
status of my library.
regards 
Kornelius

per1234, 2019-06-20:
It's fine now, that's why I removed the "Waiting for feedback" label.

One of the Arduino developers occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

KorneliusThomas, 2019-06-21:
ok. Thank you !regardsKornelius

cmaglie, 2019-06-25:
done

closed 2019-06-25 by cmaglie

KorneliusThomas, 2019-07-14:
Hi,
sorry, but I dont see my library in the Library Manager.
Please write me what is the state of this issue.
regards
Kornelius

per1234, 2019-07-14:
It is in the Arduino Library Manager index:
[image]

Please try this:

1. Open this link in your browser: http://downloads.arduino.cc/libraries/library_index.json
2. If you're using Firefox, click the "Raw Data" tab.
3. Wait for the file to load.
4. Press Ctrl + F.
5. Search the file for "PIDcontrollersModular".

Do you find an entry for "PIDcontrollersModular" in the file?

KorneliusThomas, 2019-07-14:
Yes !
I see it. Hopfully it will have success.
I have tried to put into it all my life
experience as controller developer for Arduino users.Controllers I have writtren
were 
allready standard by a very big company and runs on many plants in the world.

Best regards
Kornelius