arduino/Arduino #8969: "#include" required for ino files

varnonga, 2019-06-12:
I have a bug in the Arduino software version 1.8.9. Initially, I wrote my code to have several tabs each with some categorized functions. At first everything compiled well and there were no errors. Programs loaded into my Arduino 2560 and functioned. I made a few tweaks in my code to have some data display on a serial port LCD I had. Things were going well then, suddenly, my code stopped working. I got compile errors for some of my functions not being declared.

In order to get passed these compile errors, I had to add include statements for each tab that contained my functions. I am not sure why this happened. In one of my tabs, I had a function that called a different function under a different tab. I tried adding an #include statement for that tab in that function. I got an error about a "function definition is not allowed here before a {".  Things are really weird right now, and I don't know why.

I have an HP8570 with Windows 7 with service pack 2 installed.

per1234, 2019-06-12:
Hi @varnonga. In order for us to investigate this issue, we need to be able to reproduce it. You haven't provided enough information for us to do that. What we really need is a minimal, complete, verifiable example. This would be a complete sketch with only the minimum amount of code necessary to demonstrate the bug, and no extra unnecessary code.

PaulStoffregen, 2019-06-12:
This is very likely a mistake in your code.  The forum is probably a better place to ask.  But it may be a bug in the Arduino software... which is what this issue tracker is about.

Either way, nobody can say without seeing the complete code you used.

varnonga, 2019-06-12:
Here is my code.

Glenn Varnon
MFG Engineer  |  TechnipFMC Schilling Robotics
P  +1 530-747-2735
glenn.varnon@technipfmc.com<mailto:glenn.varnon@technipfmc.com>

This email has been sent by or on behalf of TechnipFMC plc, a company registered in England and Wales with registered no. 09909709, and with its registered office address at One St. Paul’s Churchyard, London, EC4M 8AP, or one of its subsidiaries.

 * This is where is store the error bits.
 */

byte error_bit[128] = {0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0};

/**************************************************************

pin_numbers.h: This is where we define all the pin numbers.
Written and directed by: Glenn Varnon.

****************************************************************/
#include <Arduino.h>

  
 
 //------------------------------ Pin Numbers-------------------------------
   
   //---------------communication --------------------
   
   const byte serial_rx = 19;
   const byte serial_tx = 18;      // This will be used to talk to the LCD
   
   const byte RX_1=44;            // JTAG TDO: is the input data from the test application (busy/ready)
   const byte RX_2=45;            // JTAG TDI: is the input from the test application (ready to read debug_1). Will be a logic 1 when the data on debug 1 is ready to read.
   const byte TX_1=46;            // JTAG TMS: output data pin back to the test application. Tells the test application the tester is ready or busy.
   const byte TX_2=47;            // JTAG TCK: pin go high when the data on debug_3 is ready to be read by the test application.
   const byte Force_Net_Tx = 6;
   const byte Force_Net_RX = 7;
   const byte Can_RX_Base = 8;
   const byte Can_TX_Base = 9;
   

   
   //----------------- Power ON/OFF and Isolation --------

   const byte UUT_Power_On_Off_Iso = 35;
   const byte LCD_ON_OFF  = 13;

      
   //-----------------Multiplexer Lines ----------------

   const byte AD_0 = 2;
   const byte AD_1 = 3;
   const byte AD_2 = 4;
   const byte AD_3 = 5;
   const byte AD_4 = 48;
   const byte AD_5 = 49;
   const byte AD_6 = 50;
   const byte AD_7 = 51;
   const byte Stimulas_A_Enable = 22;
   const byte Stimulas_B_Enable = 23;
   const byte Stimulas_C_Enable = 24;
   const byte Detect_A_Enable   = 25;
   const byte Detect_B_Enable   = 26;
   const byte Detect_C_Enable   = 27;
   const byte Detect_E_Enable   = 28;
   const byte Detect_F_Enable   = 29;
   const byte Detect_G_Enable   = 30;
   const byte Detect_H_Enable   = 31;
   const byte Detect_J_Enable   = 32;
    
 //--------------  Load Pins  -------------------------------------

  const byte Armnet_Load        = 36;
  const byte Tool_Load          = 37;
  const byte Camera_Load        = 38;
  const byte Armnet_26_JC_Load  = 39;
  const byte Tool_26_JC_Load    = 40;
  const byte Camera_26_JC_Load  = 41;

//--------------------- Human interface pins ---------------------
   
    const byte response_button          = 52;
    const byte troubleshoot_test_switch = 53;
    const int user_selector_switch      = A1;

//----------------   A/D ---------------------------------------------
    const int analog_measurement = A0;
    
    
     
// ------------  Future Use  and spare  -----------------------------
   
   const byte Serial_data = 20;   // Future Use
   const byte Serial_clock = 21;  // Future use

  

/*************************************************************************************************************************************************
Pass_Fail_Toleracnes: Written and directed by: Glenn Varnon
Version 0.0: Released

This is where all of the pass fail criteria reside.

*************************************************************************************************************************************************/

#include <Arduino.h>

//------------------------------------  Ground Fault Tolerances  -----------------------------------------------
const int ground_fault_limit = 250;                  // 4.096/1023 = .00400391 volts per tick; ground fault leakage should be less than 1 volt tolerance = 1.00/.0040039 = 249.7. 
                                                     // This is a 20Meg leakage limit.

//-----------------------------------   Voltage Reading tolerances  ---------------------------------------------

//-----------------------------------   Communication line tolerances  -------------------------------------------

varnonga, 2019-06-12:
Here is my code when it was still compiling. There were warnings but no errors.

Glenn Varnon
MFG Engineer  |  TechnipFMC Schilling Robotics
P  +1 530-747-2735
glenn.varnon@technipfmc.com<mailto:glenn.varnon@technipfmc.com>

This email has been sent by or on behalf of TechnipFMC plc, a company registered in England and Wales with registered no. 09909709, and with its registered office address at One St. Paul’s Churchyard, London, EC4M 8AP, or one of its subsidiaries.

matthijskooijman, 2019-06-12:
Your code does not seem to contain any functions, only constants. Perhaps your code was cut short? Also, I can't tell from your code where one file starts and the next one ends. It might be better to put your files on gist.github.com, or perhaps upload them as a zip for easy reproduction. Also be sure to include the actual error message you're getting from the IDE.

varnonga, 2019-06-12:
I guess my disconnect is the relationship between “ino” files that are created when I create a new tab. I have been using them like header files with the exception on not using an “#include” statement to keep things “connected”. It has been working quite well till now.

I think that is my main issue. At first using the “.ino” files seem to work really well. But I guess if you get too many or two “ino” files call the same “ino” files things start to fall apart.

So to clean things up, I renamed all my files with exception to my main program to “.h”. Then I used the “#include” statement to reconnect everything. Now things are functional again.

There is nothing in the documentation that really talks about the relationship between “ino” files and the “rules of engagement”. It might be nice to have an article on this somewhere.

Glenn Varnon
MFG Engineer  |  TechnipFMC Schilling Robotics
P  +1 530-747-2735
glenn.varnon@technipfmc.com<mailto:glenn.varnon@technipfmc.com>

This email has been sent by or on behalf of TechnipFMC plc, a company registered in England and Wales with registered no. 09909709, and with its registered office address at One St. Paul’s Churchyard, London, EC4M 8AP, or one of its subsidiaries.

per1234, 2019-06-12:
>There is nothing in the documentation that really talks about the relationship between “ino” files and the “rules of engagement”. It might be nice to have an article on this somewhere.

Here's the documentation on how multiple .ino files work:
https://github.com/arduino/Arduino/wiki/Build-Process#pre-processing
>All .ino files in the sketch folder (shown in the IDE as tabs with no extension) are concatenated together, starting with the file that matches the folder name followed by the others in alphabetical order

I'm not sure that information is very prominently available for a novice wondering how multiple .ino files are handled in a sketch. The path to it from arduino.cc is **Resources > Tutorials > Hacking > Build Process**. I wonder if it would make sense to also provide this information on the more obvious and prominent "Arduino Software (IDE)" page's "Tabs, Multiple Files, and Compilation" section, which currently seems quite light on information:
https://www.arduino.cc/en/Guide/Environment#toc8

PaulStoffregen, 2019-06-12:
I have no idea what these words mean:

> or two “ino” files call the same “ino” files

Files don't "call" other files.  They can "include" other files.  But the code you showed has includes only for Arduino.h.

Maybe you should find the location where Arduino is saving the files and make a ZIP archive of the folder and all the files it contains.  Then you can upload that ZIP file (perhaps a "gist" or on a file sharing site like bitbucket) so we can look at the actual files.

Also, if you ask this question on the forum (where it is almost certainly more appropriate) you will also need to clearly show all the files.  By withholding any of the files, you're making it impossible to figure out what's really wrong.  That only wastes everyone's time and hinders helping you and figuring out whether there is really any defect in the Arduino software that needs fixing.

varnonga, 2019-06-13:
Well, this explains a lot of things that have been happening to me. Wow, thanks for the information. If nothing else, it needs more exposure.

Maybe put something under “Reference”.

Glenn Varnon
MFG Engineer  |  TechnipFMC Schilling Robotics
P  +1 530-747-2735
glenn.varnon@technipfmc.com<mailto:glenn.varnon@technipfmc.com>

This email has been sent by or on behalf of TechnipFMC plc, a company registered in England and Wales with registered no. 09909709, and with its registered office address at One St. Paul’s Churchyard, London, EC4M 8AP, or one of its subsidiaries.

https://github.com/arduino/Arduino/wiki/Build-Process#pre-processing<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_arduino_Arduino_wiki_Build-2DProcess-23pre-2Dprocessing&d=DwMFaQ&c=r_B2dqKkHczsuXPCSs5DOw&r=HniJO6iP_ToibWznSmYnRzCwj0MQZb3ibTbtor8VGjU&m=U5xeTk9n62oXWwn_9OxBmzLiZQbZ-COMvLqi5hXn0og&s=3pf-gV9nJWMLfi4Y7Ap_Ob2yhEQqgGTJ5YM3SRaF8Lg&e=>

https://www.arduino.cc/en/Guide/Environment#toc8<https://urldefense.proofpoint.com/v2/url?u=https-3A__www.arduino.cc_en_Guide_Environment-23toc8&d=DwMFaQ&c=r_B2dqKkHczsuXPCSs5DOw&r=HniJO6iP_ToibWznSmYnRzCwj0MQZb3ibTbtor8VGjU&m=U5xeTk9n62oXWwn_9OxBmzLiZQbZ-COMvLqi5hXn0og&s=TtD42G19OLQp_KppVC6l41RMMZp_mQPzk1GZcHiTvto&e=>

per1234, 2019-06-13:
>Maybe put something under “Reference”.

This would be done as part of my first proposed solution to https://github.com/arduino/reference-en/issues/623
That is only a side effect of that proposal though. I don't think the Language Reference is the obvious place for someone wondering how multiple .ino tabs are handled to look.

I still think the more logical place for people using the Arduino IDE to look for information on how multiple .ino tabs are handled would be in the https://www.arduino.cc/en/Guide/Environment#toc8. However, I hadn't previously considered that users of Arduino Web Editor and arduino-cli probably don't expect to find anything of relevance in that page.

per1234, 2020-10-01:
> I'm not sure that information is very prominently available for a novice wondering how multiple .ino files are handled in a sketch. The path to it from arduino.cc is Resources > Tutorials > Hacking [...]

This part of the issue has now been resolved:
https://www.arduino.cc/en/Guide/Environment#tabs-multiple-files-and-compilation

However, the issue remains that the "Environment" documentation page is specific to the Arduino IDE, so users of other official Arduino development software will likely not think to look there. So I'll leave this open.

>I'm not sure that information is very prominently available for a novice wondering how multiple .ino files are handled in a sketch. The path to it from arduino.cc is Resources > Tutorials > Hacking > Build Process.

There is now an even longer journey to travel before finding the information via that path:
**Documentation > Tutorials > Hacking > Build Process > content moved to https://github.com/arduino/Arduino/wiki/Build-Process > content moved to https://arduino.github.io/arduino-cli/sketch-build-process/ > Pre-Processing**