arduino/Arduino #8974: Arduino IDE no longer compiles on macOS 10.15 Catalina

colinta, 2019-06-13:
Very sad!  macOS no longer supports 32bit binaries.  I deleted my copy of Arduino IDE and reinstalled the latest version, and the error message is `.../avr/bin/avr-g++: bad CPU type in executable``

facchinm, 2019-06-14:
@colinta , we are working on it, there will be a beta release with only 64 bits tools soon but not too soon (10.15 is still in dev preview)

donev77, 2019-06-14:
1.9 beta works on Catalina, tested on nodemcu ESP-12E

colinta, 2019-06-14:
Thank you all! I’ll download the beta, and test it out.

per1234, 2019-06-14:
This isn't really about the Arduino IDE. It's about the toolchain. The ESP8266 core for Arduino uses a different toolchain, which apparently is 64 bit. That's great, but not relevant here because the ESP8266 core for Arduino is a 3rd party project.

If you use the beta build of the Arduino IDE to compile for one of the AVR boards like Arduino/Genuino Uno then you'll get the same "bad CPU type in executable" error.

ttimpe, 2019-06-15:
Why is even still 32-Bit? 64-Bit has been standard for a decade.

gvarisco, 2019-06-16:
@colinta hi! could you please give build 865 a try? You can download it from https://github.com/arduino/Arduino/pull/8976

colinta, 2019-06-16:
In the latest macOS it won't let me open the app from an undientified developer, even if I override it in Security settings!  Could be a bug, there are quite a few bugs in this beta version of macOS 10.15.

gvarisco, 2019-06-16:
@colinta could you please confirm that Gatekeeper is activated on your system? What is the output of `spctl --status`?

Indeed, we do not codesign that bundle at the moment, but in case it becomes mandatory we may have to change this behaviour and update our build scripts.

colinta, 2019-06-17:
```
spctl --status
assessments enabled
...
sudo spctl --master-disable
```

And now I can run the IDE, so thanks for that heads up.  And my project compiled!  I had new errors that I didn't have in the prev IDE (around assigning `&PORTB` as a `uint8_t*` instead of `volatile uint8_t*`), but all good after that!  I haven't tried to actually flash my project yet; it's my keyboard and I want to test a quick blink project first.

colinta, 2019-06-17:
In the interest of cleaning up, I'll close this issue, if I have more problems I'll let you know!  Thanks for all the help, and everyone's work on the Arduino IDE in general.

closed 2019-06-17 by colinta

gvarisco, 2019-06-18:
Thanks for the feedback. Still, we want it to be fully working with GateKeeper turned ON, thus we'll make sure everything works as expected and without hacks before GA ships. /cc @facchinm

joztoz, 2019-07-14:
I still cant compile, still same bad cpu. tried all whats here mentioned and without success, please can you help me what I forgot install or disable/enable?
Thank you

per1234, 2019-07-14:
@joztoz please post the full error message here, including the path.

joztoz, 2019-07-15:
finally find a solution, I had to put avr files to /Users/jozefziak/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino5

but had to do it manually. Now its working fine, I take avr files from arduino 865 package :)

jakorten, 2019-10-17:
@joztoz: works really well indeed. In your console you can see one's one entire folder. I had to remove (archive) the older version though.

CallmeSk, 2019-12-12:
Same here 'bad cpu type error' while programming NodeMCUESP82466 with macOS Catalina.

JoenneSpreuwers, 2020-02-02:
https://youtu.be/xv1e6ZOBVfA <-- quick solve ;)