arduino/Arduino #8978: ATmega4809 code size exceeded without warning/error

dlkeng, 2019-06-15:
When compiling an Arduino sketch that is slightly larger than the Flash memory size of the ATmega4809, code is generated (in the hex file) past the end of Flash memory without any errors or warnings.
It appears that the "avr-size" tool is incorrectly determining the size of the program storage space.

Arduino Environment
----------------------
Arduino 1.8.8 on a Windows XP machine
Board: "Arduino Nano Every"
Registers emulation: "None (ATMEGA4809)"

Example Output:
"....\\Local Settings\\Application Data\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-size" -A "....\\LOCALS~1\\Temp\\arduino_build_140636/Test.ino.elf"
Sketch uses 48880 bytes (99%) of program storage space. Maximum is 49152 bytes.
Global variables use 290 bytes (4%) of dynamic memory, leaving 5854 bytes for local variables. Maximum is 6144 bytes.

The actual code size is: 49165 bytes (0xC00D)

End of hex file:
    :10BFFA0080000013001400280024F400C0D40100BB
    :04C00A00FF01010031
    :00000001FF

If the compiled Sketch uses more than 48874 bytes (99%), code, this problem occurs.

pnndra, 2019-06-15:
Hi,
Would you please provide the sketch that's causing the problem? Likely we can reproduce it anyway but having a common ground helps a lot

matthijskooijman, 2019-06-15:
The max size specified by the IDE is exactly 48k (49152). Assuming that this board has a bootloader like other AVR-based boards, that size should probably subtract the bootloader size (like other AVR boards do as well).

dlkeng, 2019-06-15:
No bootloader is used with the ATmega4809.

Testing sketch is below that demonstrates the problem. The actual code was run on a Microchip Curiosity Nano board since the Arduino Nano Every is not yet available. However, the problem can be seen by simply checking the end of the hex file generated. Code execution is not necessary.

In this case, the following results were shown:
"C:\\....\\Local Settings\\Application Data\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-size" -A "C:\\....\\LOCALS~1\\Temp\\arduino_build_376007/Test.ino.elf"
Sketch uses 49148 bytes (99%) of program storage space. Maximum is 49152 bytes.
Global variables use 181 bytes (2%) of dynamic memory, leaving 5963 bytes for local variables. Maximum is 6144 bytes.

End of hex file:
:10BFF800000000003C5B2A5DDC5ABA5ACE5A955BB9
:0DC00800595D945CE85A575DAB5A0D0A0073
:04C015000024F4000F
:00000001FF
-------------------------------------------------------------------
Sketch:
---------
File: Test.ino
---------
// Testing file for ATmega4809 Curiosity Nano using Arduino Nano Every target.
// This demonstrates Arduino ATmega4809 bug of not detecting full Flash space!

#include "test.h"

// Debug USART
//  Default Arduino "Serial" mapped to USART3 on alternate pins PB4/PB5 [D27]/[D26]
//  For Curiosity Nano, "Serial" needs re-mapped to USART3 on PB0/PB1 [D9]/[D10]
//----------------------------------------------------------------------

#define CURIOSITY_LED_PIN   3   // Arduino Dx pin - PF5
#define THE_LED_PIN         CURIOSITY_LED_PIN

#define BAUD_RATE           9600        // actual baud rate
#define CLK_CORR_FACTOR     (20.0 / 16.0)   // adjustment due to MCU running at 20Mhz instead of expected 16Mhz
#define THE_BAUD_RATE       (BAUD_RATE / CLK_CORR_FACTOR)

#define PB0_PIN     9       // Arduino Dx pin - PB0
#define PB1_PIN     10      // Arduino Dx pin - PB1

const uint8_t ledPin = THE_LED_PIN;     // the number of the LED pin
uint32_t Interval = 250;                // interval at which to blink (milliseconds)

PinStatus ledState = LOW;           // ledState used to set the LED

void setup(void) 
{
    // initialize digital pin ledPin as an output.
    pinMode(ledPin, OUTPUT);

    // initialize serial communications
    Serial.begin(THE_BAUD_RATE);

    // remap the pins (undo the Arduino Nano Every pin mappings)
    pinMode(PB0_PIN, OUTPUT);
    pinMode(PB1_PIN, INPUT);
	PORTMUX.USARTROUTEA &= ~PORTMUX_USART3_gm;
   
    Serial.println(TestData1[0]);
    Serial.println(TestData2[0]);
    Serial.println(TestData3[0]);
    Serial.println(TestData4[0]);
    Serial.println(TestData5[0]);
}

void loop(void) 
{
    static uint32_t previousMillis = 0;   // will store last time LED was updated
    uint32_t currentMillis = millis();

    // check to see if it's time to blink the LED
[... 589 lines cut ...]
    "This is a test126: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test127: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test128: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test129: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test130: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test131: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test132: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test133: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test134: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test135: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test136: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test137: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test138: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test139: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test140: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test141: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test142: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
};
const char TestData5[][80] PROGMEM =
{
    "This is a test000: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test001: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
    "This is a test002: ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz",
};
---------

facchinm, 2019-06-16:
Hi @dlkeng,
that's totally my fault. Version 1.6.x of the megaAVR core didn't contain any bootloader while 1.8.1 does (starting from https://github.com/arduino/ArduinoCore-megaavr/commit/a33109d949d969cae04e64c60a71282cce68a9bc#diff-d638a47d1ca0e2a5585fb34c7d7fb0f9R24 ).
So I forgot to remove that 0x200 bytes from `uno2018.upload.maximum_size` field; in the next few days I'll produce an updated core for megaAVR

closed 2019-06-16 by facchinm