arduino/Arduino #8979: Windows 10 java.io.IOException: Cannot run program "{runtime.tools.bossac-1.7.0.path}

robynjayqueerie, 2019-06-15:
This is close to previous issues but I can't resolve it using the methods described there.
I have everything working under Windows 7, am trying to install on a new Windows 10 machine.
Using an Adafruit Feather M0 device and get the following error when I attempt to upload the compiled sketch
java.io.IOException: Cannot run program "{runtime.tools.bossac-1.7.0.path}/bossac": CreateProcess error=2, The system cannot find the file specified + a heap else
I use the same board and cable on my Win 7 installation, all OK
Am using the latest Arduino IDE 1.8.9
I have completely uninstalled the IDE several times, deleted the Arduino15 folder in Users/AppData etc, installed the Adafruit and Arduino SAMD boards as per instructions and as per the Win7 installation.
I can get an ESP32 compile and upload to happen (selecting appropriate boards of course)
The only issue with Win10 I observed was that it does not seem to install Java by default, I had to do that separately (and the Arduino install is happening after that) possibly I have not installed the correct version although it was the latest available.
I have disabled virus checking (Malwarebytes stopped completely - no other virus checking that I am aware of unless Win10 is doing something behind the scenes)
Any help would be very welcome, thanks in advance

per1234, 2019-06-15:
This is a bug on Adafruit's end:
https://github.com/adafruit/arduino-board-index/issues/22
for which a pull request has been submitted (but not yet merged):
https://github.com/adafruit/arduino-board-index/pull/23/files

per1234, 2019-06-15:
The workaround is to install Arduino SAMD Boards 1.6.21 in Boards Manager.

closed 2019-06-15 by per1234

robynjayqueerie, 2019-06-15:
What amazing service on a Saturday afternoon! Thanks, installing SAMD version 1.6.21 instead of the latest version fixed it
Happiness descends like a dove upon my head and I can sleep the dreams of contentment once more!