arduino/Arduino #8982: ESPRESSIF ESP Development Board WIFi Cache Cannot Clear

mxlbb, 2019-06-16:
Cache cannot be cleared using "Wireless Scan" and "Smart Distribution Network" in the Arduino sample project.
Performance: If you use these two functions before the wireless LAN is not connected, then WiFi multiple scans will be the same result. Only when the wireless network is restarted or disconnected again can the scan result be cleared.
The BUG has not been debugged using the ESPRESSIF development tool.
I hope it can be solved as soon as possible. Thank you.

per1234, 2019-06-16:
The ESP8266 core for Arduino is a 3rd party project. Arduino's issue trackers are only to be used to report bugs or make feature requests for Arduino's own projects.

closed 2019-06-16 by per1234

mxlbb, 2019-06-16:
Who should I look for?