arduino/Arduino #8988: package_index.json file signature verification failure for some users

per1234, 2019-06-17:
We have had four reports of this error on the forum in the last two days. The usual trick of clearing out the data folder and trying again didn't work for any of them. One of the users, Cheetor, provided the package_index.json and package_index.json.sig files they get from https://downloads.arduino.cc/packages/package_index.json and https://downloads.arduino.cc/packages/package_index.json.sig:
- https://forum.arduino.cc/index.php?topic=621811.msg4212477#msg4212477
- https://forum.arduino.cc/index.php?action=dlattach;topic=621811.0;attach=312913

I compared these to the files I download from the same URLs and found that their package_index.json file was missing the entries for Arduino SAMD Boards 1.8.1 and avrdude 6.3.0-arduino17, but no differences other than that. The checksum of their .sig file matches mine.

Cheetor is in New Zealand and one of the other reporting users (DavidBMason) is as well. The other two haven't provided their location. The problem stopped occurring for DavidBMason before I could get the bad package_index.json and package_index.json.sig files from them:
https://forum.arduino.cc/index.php?topic=621637.msg4212584#msg4212584

My hypothesis is that there was a recent update to package_index.json but the new .json file didn't make it to a server that provides the files to people in NZ. However, the new .sig file did make it to that server. So they are getting the old .json file but the new .sig file, thus the signature verification. Further evidence of this is that when Cheetor used TOR with an exit node in the USA they got the new version of package_index.json:
https://forum.arduino.cc/index.php?topic=621811.msg4212512#msg4212512

It would be nice if there was some way to make sure that the .json and .sig files will always hit the servers at the same time. I suspect this delay of days on the .json file is a rare glitch but if we regularly have a delay of even minutes that still is going to cause problems for people, more so because of https://github.com/arduino/Arduino/issues/8936.

Forum threads:
- https://forum.arduino.cc/index.php?topic=621497
- https://forum.arduino.cc/index.php?topic=621637
- https://forum.arduino.cc/index.php?topic=621811

facchinm, 2019-06-17:
@rsora or @endorama may you take a look?

endorama, 2019-06-17:
@rsora will take care on this. I'm subscribed to the issue, so just ping if you need me!

rsora, 2019-06-17:
I'm reviewing the issue and opening an internal Incident report, I'll keep you posted!

ilmarhrundel, 2019-06-17:
I have the same problem on two PCs and four different internet connections..

hortynz, 2019-06-17:
I am in NZ too, and I also had the same problem on two different PCs across several web connections.  I didn't try tunnelling yet - would have to set something up.  I'm still experiencing the issue this morning, and have attached the two files I get.
[package_index.json.txt](https://github.com/arduino/Arduino/files/3298281/package_index.json.txt)
[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3298282/package_index.json.sig.txt)

hortynz, 2019-06-17:
not sure if searches will scan the text of our issue, but just in case, I'm pasting the error message displayed on my sketch UI.  "package_index.json file signature verification failed. File ignored"

per1234, 2019-06-17:
Thanks for sharing those files @hortynz! They have the same issue as the files shared on the forum by Cheetor. package_index.json is missing the entries for Arduino SAMD Boards 1.8.1 and avrdude 6.3.0-arduino17 that are present in my download from https://downloads.arduino.cc/packages/package_index.json but the checksum of the .sig file matches the checksum of the .sig file I get from https://downloads.arduino.cc/packages/package_index.json.sig.

rsora, 2019-06-18:
Hi there,
Can someone provide the http headers that are received when calling both
https://downloads.arduino.cc/packages/package_index.json https://downloads.arduino.cc/packages/package_index.json.sig
in case you are still experiencing the signature error? 
Thanks!

anzas, 2019-06-18:
Here are the http headers
[package_index_json_header.txt](https://github.com/arduino/Arduino/files/3300952/package_index_json_header.txt)
[package_index_json_sig_header.txt](https://github.com/arduino/Arduino/files/3300953/package_index_json_sig_header.txt)

rsora, 2019-06-18:
Hi there,
I' have just launched a CDN refresh can you please verify if you are still experiencing the
 _ "package_index.json file signature verification failed. File ignored"_ 
error? 
If yes please reply with geographical location and both headers and files.
Thanks!
cc @hortynz @per1234 @anzas

anzas, 2019-06-18:
That fixed it, at least for me. Location is Finland.
[header_json.txt](https://github.com/arduino/Arduino/files/3301799/header_json.txt)
[header_json_sig.txt](https://github.com/arduino/Arduino/files/3301800/header_json_sig.txt)
[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3301801/package_index.json.sig.txt)
[package_index.json.txt](https://github.com/arduino/Arduino/files/3301802/package_index.json.txt)

ilmarhrundel, 2019-06-18:
Everything is working now! (Estonia)
Thank You very much.

hortynz, 2019-06-18:
here too, thanks

cheetor5923, 2019-06-18:
Can confirm, CDN refresh has worked. I'm now getting the correct file

bhavanakrishna, 2019-06-19:
http://arduino.esp8266.com/stable/package_esp8266com_index.json file signature verification failed. File ignored. 
help me out with this error 
i tried different versions of arduino ide and different  system but same error

rsora, 2019-06-20:
Hi @bhavanakrishna,
the _package_esp8266com_index.json_ is a third party index that is not served by our services, I suggest you to ask in the https://www.esp8266.com/ forum for hints on how to solve your problem!

Regarding this issue, having received a positive feedback both in the forum and in the previous message, I'll proceed to close and solve this issue.

Thanks to @hortynz @per1234 @anzas @cheetor5923 @ilmarhrundel for providing feedback!

BZ840, 2019-06-20:
Hi @rsora,
I encountered the same issue on multiple devices since yesterday: https://downloads.arduino.cc/packages/package_index.json file signature verification failed. File ignored.
Here are the files:
[package_index.json.txt](https://github.com/arduino/Arduino/files/3310991/package_index.json.txt)

Thank you for your help in advance!

endorama, 2019-06-21:
Hi @BZ840, please provide the HTTP headers you get when requesting that file, so we can look into it!

BZ840, 2019-06-21:
Hi @endorama,
Here is the file 
[http_header.txt](https://github.com/arduino/Arduino/files/3314759/http_header.txt)

Thanks!

endorama, 2019-06-21:
Hi @BZ840 sorry for not being clearer, but we need both `.json` and `.json.sig` requests headers to debug this.

As reference take [this](https://github.com/arduino/Arduino/issues/8988#issuecomment-503147453) previous comment. 

May you also share which nation are you connecting from?

Thanks

BZ840, 2019-06-21:
Hi @endorama 
[header_json.txt](https://github.com/arduino/Arduino/files/3315191/header_json.txt)
[header_json.sig.txt](https://github.com/arduino/Arduino/files/3315192/header_json.sig.txt)
The location is Canada.

CLK88, 2019-06-22:
I have the same problem and I'm from Canada. 

https://downloads.arduino.cc/packages/package_index.json
https://downloads.arduino.cc/packages/package_index.json.sig 

It's causing all MKR boards to not be listed and I can't search them in my boards manager.

red-scorp, 2019-06-23:
Same problem
OS: Windows
Location: Germany
OS Language: US

[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3317577/package_index.json.sig.txt)
[package_index.json.txt](https://github.com/arduino/Arduino/files/3317578/package_index.json.txt)

How can I grab the http header?
How can I  launched a CDN refresh?
Thanks in advance!

per1234, 2019-06-24:
>How can I grab the http header?

1. Open a new browser tab or window.
2. Press F12 to open the toolbox.
3. Click the "Network" tab of the toolbox.
4. On the next bar down in the toolbox, click "All".
5. Paste the URL (https://downloads.arduino.cc/packages/package_index.json or https://downloads.arduino.cc/packages/package_index.json.sig) into the URL bar of your browser.
6. Press Enter.
7. In the toolbox, click on the line that says "package_index.json" or "package_index.json.sig" (depending on which URL you're currently getting the headers for).
8. In the pane that appears, click the "Headers" tab.
9. If using Firefox, switch the "Raw headers" switch to the on position for the "Response headers" section.
10. Click and drag to select all text in the "Response headers" section.
11. Press Ctrl + C to copy the selected text to the clipboard.
12. You can now either paste the copied header text directly into a reply here or save it in a .txt file and attach the .txt file in a reply here.
13. Repeat the process for the other URL.

BZ840, 2019-06-24:
Hi @endorama 
Those two header files I provided 3 days ago, were used some http header online service to retrieve by pasting the URL.
Here is what I got if I follow what @per1234 suggested:
[header.json.txt](https://github.com/arduino/Arduino/files/3321386/header.json.txt)
[header.json.sig.txt](https://github.com/arduino/Arduino/files/3321387/header.json.sig.txt)
The location is Canada.

I apologize for any confusion and inconvenience.

red-scorp, 2019-06-24:
https://downloads.arduino.cc/packages/package_index.json
```
Cache-Control: public, max-age=2419200
CF-Cache-Status: HIT
CF-Ray: 4ec183b538dd7cd6-MUC
Connection: keep-alive
Date: Mon, 24 Jun 2019 20:40:02 GMT
ETag: "266a3f9e6b360886220fc1815d356bfe"
Expect-CT: max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
Expires: Mon, 22 Jul 2019 20:40:02 GMT
Last-Modified: Fri, 24 May 2019 13:52:41 GMT
Server: cloudflare
Vary: Accept-Encoding
Via: 1.1 acc9aed747aea07d6138203ddfb2dcd9.cloudfront.net (CloudFront)
X-Amz-Cf-Id: 6WmU3rql1m5EWvvQmQtbFw5hnIqpSx8W49T9zwQ-XWqJ6TPNYxL-pw==
x-amz-replication-status: COMPLETED
x-amz-version-id: aujWQVWH2.tNQmcPhdSJiy.4gqfqDncA
X-Cache: RefreshHit from cloudfront

```

red-scorp, 2019-06-24:
https://downloads.arduino.cc/packages/package_index.json.sig
```
Accept-Ranges: bytes
Cache-Control: public, max-age=2419200
CF-Cache-Status: HIT
CF-Ray: 4ec17f72e94f7cd6-MUC
Content-Length: 543
Content-Type: application/pgp-signature
Date: Mon, 24 Jun 2019 20:37:07 GMT
ETag: "0023b680091faf0064985b79da7020a7"
Expect-CT: max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
Expires: Mon, 22 Jul 2019 20:37:07 GMT
Last-Modified: Wed, 19 Jun 2019 15:10:02 GMT
Server: cloudflare
Vary: Accept-Encoding
Via: 1.1 81f038b63d8af92c2b360530d51919c2.cloudfront.net (CloudFront)
X-Amz-Cf-Id: aeZ3vmVN7ICrQWltZ8eqJmvNkgPl5M2lEdoOCs3YynppQWBJ3vorkw==
X-Amz-Cf-Pop: MUC50-C1
x-amz-replication-status: COMPLETED
x-amz-version-id: g4dBS._ndEMgrPdvRNS9v.vv7ZBbCxJX
X-Cache: RefreshHit from cloudfront
```

red-scorp, 2019-06-24:
As I mentioned earlier, location is Germany.
The same happened on another PC in different network.
I can provide readers for this PC also if you want.

rsora, 2019-06-25:
Thanks for the heads up, I'm re-opening the issue and start the investigations.
I'll give you all  updates once available.

rsora, 2019-06-25:
I tried to replicate your problems using the files and headers you provided but I have inconsistent behaviors:

@BZ840:
- the headers provided from the "online servers" are correct and updated, while the headers you obtained using the @per1234 steps reports an outdated `.sig` file
- anyway I tried manually do check differences and signature for the `package.json` file you shared and both signature and last update check are ok

@red-scorp 
- I see from the headers that you have an old `package.json` file and an updated signature
- both the file you shared seem updated and the signature check i performed locally on my pc is ok too

@CLK88 
- I have not sufficient information in order to say how to solve your problem

Having these behaviors, I ask you all to follow the guide below in order to help you to both solving your issues, or in the worst case, obtain all the information to assess properly the problem our side.

> ### Step A
> 1. (In the Arduino IDE) File > Preferences
> 2. Click the link at the line following "More preferences can be edited directly in the file". This will open the `Arduino15` (or similar name depending on OS) folder.
> 3. Delete all files in that folder except for preferences.txt. Please be very careful when deleting things on your computer. When in doubt, back up!
> 4. (In the Arduino IDE) Close the Boards Manager window if it's open.
> 5. Tools > Board > Boards Manager
> 6. Wait for the downloads to finish.
> 
> 
> Does the "file signature verification failed" error still occur?
> 
> **No**
> It was likely caused by a temporary glitch in the Arduino IDE and should now be fixed.
> 
> **Yes**
> Go on to the next step:
> 
> ### Step B
> 1. Close all Arduino IDE windows.
> 2. If using Windows, download the "Windows ZIP file for non admin install" version of the Arduino IDE from the [Software page](https://www.arduino.cc/en/Main/Software). Unzip the downloaded file to any convenient location on your computer where you have write access. Don't put it under `C:\Program Files` or `C:\Program Files (x86)` because Windows 10 places extra restrictions on those folders.
> 3. Create a folder named `portable` in the Arduino IDE installation folder to set the Arduino IDE into [portable mode](https://www.arduino.cc/en/Guide/PortableIDE). If you are using macOS, you can follow [these instructions](https://www.arduino.cc/en/Guide/PortableIDE#toc3).
> 4. Start the Arduino IDE.
> 5. Tools > Board > Boards Manager
> 6. Wait for the downloads to finish.
> 
> Does the "file signature verification failed" error still occur?
> 
> **No**
> It may have been caused by your antivirus software being more restrictive of the location of the Arduino15 folder. You may be able to fix the issue by adjusting your antivirus software's settings.
> 
> **Yes**
> Go on to the next step:
> 
> ### Step C
> 1. **TEMPORARILY** disable your antivirus software.
> 2. (In the Arduino IDE) Close the Boards Manager window if it's open.
> 3. Tools > Board > Boards Manager
> 4. Wait for the downloads to finish.
> 5. Re-enable your antivirus software immediately.
> 
> Does the "file signature verification failed" error still occur?
> 
> **No**
> You'll need to whitelist the appropriate folder, file, or process in your antivirus software's settings.
> 
> **Yes**
> Go on to the next step:
> 
> ### Step D
> 1. Connect to the Internet through a different network. If you're currently using the network at your work, try your home network. You can try connecting via the WiFi hotspot on your phone.
> 2. (In the Arduino IDE) Close the Boards Manager window if it's open.
> 3. Tools > Board > Boards Manager
> 4. Wait for the downloads to finish.
> 
> Does the "file signature verification failed" error still occur?
> 
> **No**
> The issue may be caused by the firewall on the original network.
> 
> **Yes**
> Go on to the next step:
> 
> ### Step E
> 1. Open a new browser tab or window.
> 2. Press F12 to open the toolbox.
> 3. Click the "Network" tab of the toolbox.
> 4. On the next bar down in the toolbox, click "All".
> 5. Check the box next to "Disable cache".
> 5. Paste the URL (https://downloads.arduino.cc/packages/package_index.json or https://downloads.arduino.cc/packages/package_index.json.sig) into the URL bar of your browser.
> 6. Press Enter.
> 7. In the toolbox, click on the line that says "package_index.json" or "package_index.json.sig" (depending on which URL you're currently getting the headers for).
> 8. In the pane that appears, click the "Headers" tab.
> 9. If using Firefox, switch the "Raw headers" switch to the on position for the "Response headers" section.
> 10. Click and drag to select all text in the "Response headers" section.
> 11. Press Ctrl + C to copy the selected text to the clipboard.
> 12. You can now either paste the copied header text directly into a reply here or save it in a .txt file and attach the .txt file in a reply here.
> 13. In the browser window where you have the package_index file open, press Ctrl + S
> 14. Save the file.
> 15. Attach the saved file to your reply.
> 13. Repeat the process for the other URL.

Thanks for your time!

red-scorp, 2019-06-25:
@rsora Based on what you said, this is a network/server problem, isn't it?

I made steps A and B, it did not help. I can not perform steps C and D because I have no admin permissions on this PC.

What can I do to make Boards Manager work?

Thanks in advance!

BZ840, 2019-06-25:
Hi @rsora,
The issue is now resolved, it is caused by the university wifi. 
Thank you so much!

hm-heli, 2019-06-26:
hi, does still not work for me in Germany.
I did steps A through D.
If you wish I can provide step E
Regrads, Hermann

gdeflaux, 2019-06-26:
I've had the same issue from South Africa. Connecting to internet via a VPN in France solved it.

TheTrueForce, 2019-06-27:
I'm having the same issue in Australia.
What made mine all the weirder was that I'd literally just compiled that sketch for the Due, changed it to the Uno, and then couldn't switch back to the Due.

A: The verification failed. No downloads began
B: Same as A. Verification failed, no downloads were started.
C: Same as A.
D is not an option for me. I do not have access to another internet connection.
E: The JSON's response header:
`HTTP/1.1 200 OK
Date: Thu, 27 Jun 2019 01:47:36 GMT
Content-Type: application/json
Transfer-Encoding: chunked
Connection: keep-alive
CF-Cache-Status: HIT
Cache-Control: public, max-age=2419200
CF-Ray: 4ed3c1013876da46-SYD
Age: 642972
ETag: W/"eaf07cd550c0f00f2d10e944f58a6356"
Expect-CT: max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
Expires: Thu, 25 Jul 2019 01:47:36 GMT
Last-Modified: Wed, 19 Jun 2019 15:10:01 GMT
Vary: Accept-Encoding
Via: 1.1 30f80e470abaf8292272e872df09bc50.cloudfront.net (CloudFront)
X-Amz-Cf-Id: XrEdxqfeX65vZ00G1Hozcqa_oqjRXlzUAWLLW0S_PDGLFoxUkNINqQ==
X-Amz-Cf-Pop: SFO5-C3
x-amz-replication-status: COMPLETED
x-amz-version-id: KMuAw10TOM8GHmMIU6x._J5N4wt59qdM
X-Cache: RefreshHit from cloudfront
Server: cloudflare
Content-Encoding: gzip`
SIG's response header:
`HTTP/1.1 200 OK
Date: Thu, 27 Jun 2019 01:50:27 GMT
Content-Type: application/pgp-signature
Content-Length: 543
Connection: keep-alive
CF-Cache-Status: HIT
Cache-Control: public, max-age=2419200
CF-Ray: 4ed3c52d6bbfda46-SYD
Accept-Ranges: bytes
Age: 1405578
ETag: "279cf9c5a421b4b9fa57e7824c426788"
Expect-CT: max-age=604800, report-uri="https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct"
Expires: Thu, 25 Jul 2019 01:50:27 GMT
Last-Modified: Wed, 05 Jun 2019 15:59:23 GMT
Vary: Accept-Encoding
Via: 1.1 42f9f0e9bd0296c3bb45648019b2dce5.cloudfront.net (CloudFront)
X-Amz-Cf-Id: IDv17AwhqQZ6-0ZENHzL_oEDJ9g-uckRfYy2Fz9d4UlBgLHEIwyW8A==
x-amz-replication-status: COMPLETED
x-amz-version-id: mR_9xNlB.01OMvTT8PzaVrHZk18.YA0K
X-Cache: Miss from cloudfront
Server: cloudflare`

per1234, 2019-06-27:
Thanks @TheTrueForce!

Please also attach the package_index.json and package_index.json.sig files in a reply here. You'll need to save these files with a .txt extension so that GitHub will allow them to be attached.

TheTrueForce, 2019-06-27:
I have done a couple of refreshes since those logs, first checking libraries and then boards.
The sig file is not present in my Arduino15 directory. This may be a problem.

[library_index.json.txt](https://github.com/arduino/Arduino/files/3334520/library_index.json.txt)

red-scorp, 2019-06-27:
[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3334561/package_index.json.sig.txt)
[package_index.json.txt](https://github.com/arduino/Arduino/files/3334562/package_index.json.txt)
[header.json.sig.txt](https://github.com/arduino/Arduino/files/3334563/header.json.sig.txt)
[header.json.txt](https://github.com/arduino/Arduino/files/3334564/header.json.txt)

Again from another PC in Germany... Same issue

cfulton, 2019-06-27:
Also having the issue from Australia from about 16 hours ago.
[package_index.json.txt](https://github.com/arduino/Arduino/files/3336980/package_index.json.txt)
[package_index.json.header.txt](https://github.com/arduino/Arduino/files/3336981/package_index.json.header.txt)
[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3336982/package_index.json.sig.txt)
[package_index.json.sig.header.txt](https://github.com/arduino/Arduino/files/3336983/package_index.json.sig.header.txt)

chrisoutdoorwork, 2019-06-28:
Another issue from Sydney Australia

[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3337284/package_index.json.sig.txt)
[package_index.json.sig_headers.txt](https://github.com/arduino/Arduino/files/3337285/package_index.json.sig_headers.txt)
[package_index.json.txt](https://github.com/arduino/Arduino/files/3337286/package_index.json.txt)
[package_index.json_headers.txt](https://github.com/arduino/Arduino/files/3337287/package_index.json_headers.txt)

cfulton, 2019-06-28:
Using a VPN works. So I guess thats a solution for anyone in Australia ATM.

hm-heli, 2019-06-28:
hi,
I'm using 1.8.5 because my Nanos are not uploaded with newer verisons.
When I quit the IDE, package_index.json and package_index.json.sig are deleted from Arduino15.
When I copy these two files back from a backup I made earlier, all works fine again, 'till I quit the IDE.
Therefore I face actually 2 Problems (in my case):
The IDE deletes the jsons and  when restarted can't reload them from internet with above mentioned error msg.
Regards, Hermann

rsora, 2019-06-28:
Hi there,

hm-heli, 2019-06-28:
hi, nothing has changed for me.
since I don't get the json files after all, I can't provide them
(Unless your are interested in my working backup jsons).
Regards, Hermann

rsora, 2019-06-28:
@hm-heli, can you follow the step E of the guide posted in my previous comment and upload the files you get and the related http headers? Thanks!

hm-heli, 2019-06-28:
hi, I hope i have it right...
[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3339997/package_index.json.sig.txt)
[package_index.json.txt](https://github.com/arduino/Arduino/files/3339998/package_index.json.txt)

per1234, 2019-06-28:
@hm-heli those are the headers, which is great, but we also need you to provide the actual files you download from the URLs:

- https://downloads.arduino.cc/packages/package_index.json
- https://downloads.arduino.cc/packages/package_index.json.sig

hm-heli, 2019-06-29:
hi, I hope I got it right this time,
although I'm a little confused that I can download
the files with the browser but not with the IDE :-)
Regards, Hermann
[header_json.txt](https://github.com/arduino/Arduino/files/3341668/header_json.txt)
[header_json_sig.txt](https://github.com/arduino/Arduino/files/3341669/header_json_sig.txt)
[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3341670/package_index.json.sig.txt)
[package_index.json.txt](https://github.com/arduino/Arduino/files/3341671/package_index.json.txt)

pdo-smith, 2019-06-29:
I have the same problem.
South Africa, using Ubuntu 19.04
[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3342316/package_index.json.sig.txt)
[package_index.json.txt](https://github.com/arduino/Arduino/files/3342317/package_index.json.txt)
[header.json.txt](https://github.com/arduino/Arduino/files/3342327/header.json.txt)
[header-json.sig.txt](https://github.com/arduino/Arduino/files/3342329/header-json.sig.txt)

red-scorp, 2019-06-30:
@rsora It's suddenly works today. I've tried several times after you triggered a content invalidation.
At least Arduino IDE sees all the boards now.
Thanks!

pdo-smith, 2019-06-30:
_file signature verification failed. File ignored_

Sadly the problem still exists for me this morning. I see no progress. My Arduino Due sits helplessly on my desk, begging for some code!

Is it not possible to implement some manual work around in the mean time?

bassai-nx01, 2019-06-30:
here is my data:

[package_index.json.info.txt](https://github.com/arduino/Arduino/files/3342680/package_index.json.info.txt)
[package_index.json.sig.info.txt](https://github.com/arduino/Arduino/files/3342681/package_index.json.sig.info.txt)
[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3342682/package_index.json.sig.txt)
[package_index.json.txt](https://github.com/arduino/Arduino/files/3342683/package_index.json.txt)

red-scorp, 2019-06-30:
@rsora Now it does not work again..

[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3342724/package_index.json.sig.txt)
[package_index.json.txt](https://github.com/arduino/Arduino/files/3342725/package_index.json.txt)
[header.json.sig.txt](https://github.com/arduino/Arduino/files/3342726/header.json.sig.txt)
[header.json.txt](https://github.com/arduino/Arduino/files/3342727/header.json.txt)

What also come to my mind is, I have third party packages installed. Can This be a reason for this behavior in some way?

[image]

hm-heli, 2019-06-30:
@red-scorp:
I don't have any third party software installed.

@pdo-smith:
I've commented this 2 days ago:
I'm using 1.8.5 because my Nanos are not uploaded with newer verisons.
When I quit the IDE, package_index.json and package_index.json.sig are deleted from Arduino15.
When I copy these two files back from a backup I made earlier, all works fine again, 'till I quit the IDE.
Therefore I face actually 2 Problems (in my case):
The IDE deletes the jsons and when restarted can't reload them from internet with above mentioned error msg.

chrisoutdoorwork, 2019-07-01:
Previously i tried through two different networks: one through a PFSense router that i manage, and the other through the work domain managed network.
This morning my laptop did a windows update before i could test the Arduino software again
I am now able to do the update and gain access to the correct board again. (arduino uno wifi rev2). This was done through the PFSense router network

[package_index.json.sig.txt](https://github.com/arduino/Arduino/files/3343497/package_index.json.sig.txt)
[package_index.json.sig_headers.txt](https://github.com/arduino/Arduino/files/3343498/package_index.json.sig_headers.txt)
[package_index.json.txt](https://github.com/arduino/Arduino/files/3343499/package_index.json.txt)
[package_index.json_headers.txt](https://github.com/arduino/Arduino/files/3343500/package_index.json_headers.txt)

TheTrueForce, 2019-07-01:
It no longer errors for me.

pdo-smith, 2019-07-01:
Even when I download package_index.json.sig and place it in the .arduino15 folder I still get the error. So a manual workaround does not seem possible for me.

pdo-smith, 2019-07-01:
I m happy to report that it now works.

pdo-smith, 2019-07-01:
And now it fails again, exactly as it did before.

Ranothil, 2019-07-01:
still failing

[52 more replies]