arduino/Arduino #8990: My Library has disappeared from Library Manager

iliaslamprou, 2019-06-17:
Hello,
I can't find my library on Arduino Library Manager any more
I have not made any changes
The library name is: Virtuino
The library link is: https://github.com/iliaslamprou/virtuino
It happens after the installing the latest version 1.8.9
Thank you in advance
Regards

per1234, 2019-06-17:
I don't think it was ever added in the first place. You submitted a request to add your virtuino libraries here:
https://github.com/arduino/Arduino/issues/8882
but then closed the request before they were added to the Library Manager index and opened a new request at the same time:
https://github.com/arduino/Arduino/issues/8901
but that request was only for iliaslamprou/virtuino_stm32, which was added and is still available from Arduino Library Manager.

Would you like us to also add the other library you requested in https://github.com/arduino/Arduino/issues/8882: https://github.com/iliaslamprou/virtuinoESP?

iliaslamprou, 2019-06-18:
Thanks for the reply,
I have 3 libraries.  
https://github.com/iliaslamprou/virtuinoESP
https://github.com/iliaslamprou/virtuino_stm32
https://github.com/iliaslamprou/virtuino

Two of them are already on the Library manager. 
I want to add and the third library: https://github.com/iliaslamprou/virtuino
What I have to do now? Could you add it?
Thank you in advance

Two of then are already on the

PaulStoffregen, 2019-06-18:
While not technically required, you might also consider improving the description ("sentence" and "paragraph" fields) in your library.properties file.  Currently it says only "Required for communication with Virtuino app".  I suppose if everyone who chooses you use your library already has this app installed, then this makes sense.  But by not offering any description of the app, not saying what the combination of your app and your library actually does, what value it offers, you're probably missing an opportunity to gain more users for both your library and app.

iliaslamprou, 2019-06-18:
Thanks for your suggestion. I have changed the information link to help the users of the library.
The truth is that the library is useful for users that have already installed the app.
Finally what I have to do now to add the library to the Library Manager?

per1234, 2019-06-18:
>what I have to do now to add the library to the Library Manager?

No further action is required from you. One of the Arduino developers periodically goes through all the outstanding requests for inclusion in the Arduino Library Manager index and adds them all. They will close this issue at that time.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

cmaglie, 2019-06-25:
done

closed 2019-06-25 by cmaglie