arduino/Arduino #8992: [Library Manager] Please add SimpleRotary

mprograms, 2019-06-18:
SimpleRotary is a Arduino library for using basic bare-bones type rotary encoders. Usually these are the cheap ones without the breakout board.

Supports 
- Getting rotary encoder rotation direction.
 - Getting rotary select button press.
 - Getting rotary select button long press.
 - Getting rotary select button press time.
 - Basic debouncing of rotary / button.
 - Basic error correcting of out-of-sync rotational readings.
 - Use of both high and low triggers.

https://github.com/mprograms/SimpleRotary

per1234, 2019-06-18:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

mprograms, 2019-06-18:
All done.

cmaglie, 2019-06-25:
done

closed 2019-06-25 by cmaglie