arduino/Arduino #8993: Java bundled with Arduino 1.8.9 broken on Ubuntu 18.04.2 LTS due to missing libjli

luigicalligaris, 2019-06-18:
When running the Arduino IDE 1.8.9 on one of my machines (Kubuntu 18.04.2 LTS), the bundled java interpreter throws a linker error, failing to find libjli.so

> $ ./arduino-1.8.9/arduino
> /home/user/opt/arduino-1.8.9/java/bin/java: error while loading shared libraries: libjli.so: cannot open shared object file: No such file or directory

This behaviour is not observed with the previous version 1.8.8, which starts up correctly. 

> $ ./arduino-1.8.8/arduino
> Picked up JAVA_TOOL_OPTIONS:

This previous version bundles libjli, while 1.8.9 seems to be lacking this library. The library sits at:

`./java/lib/amd64/jli/libjli.so`

I downloaded 1.8.9 from the official site as a tgz archive for Linux amd64 machines.

facchinm, 2019-06-19:
Hi @luigicalligaris ,
I just tested 1.8.9 on a fresh Ubuntu 18.04.2 installation and the IDE starts well.
I think something may have failed when you untarred the 1.8.9 package. Whould you mind removing the folder and trying again? Thanks!

luigicalligaris, 2019-06-19:
Hi Martino @facchinm, I re-downloaded the archive from the site, untarred again and it appears that this archive had more files in it. Weird, I don't recall seeing errors related to a premature end of the tar archive. Now the library is there, and it works.

closed 2019-06-19 by luigicalligaris

PanderMusubi, 2019-07-14:
Please, see also https://bugs.launchpad.net/ubuntu/+source/arduino/+bug/1425677

per1234, 2019-07-14:
@PanderMusubi the bug report you linked is about the version of the Arduino IDE in the Ubuntu package repository not being able to be updated past Arduino IDE 1.0.5 due to licensing problems with the IDE source code (https://github.com/arduino/Arduino/pull/2703). I don't see how that's in any way relevant to this issue.

PanderMusubi, 2019-07-15:
Was more that that issue also is also related to Ubuntu and Java, not specifically only the license issue and version update. As there are not that many people able to help with the packaging of Arduino for Ubuntu, I mentioned it here in the hope to find some more help, nothing more than that. Sorry if it caused some confusion.