arduino/Arduino #8998: [Library Manager] Please add Ambient library

TakehikoShimojima, 2019-06-20:
We created a library to send IoT data to a cloud service named "Ambient".
Ambient is a cloud service to receive IoT data, store them, and visualize them.
Please add this library to Library Manager.

URL: https://ambidata.io

Thank you.

per1234, 2019-06-20:
Is this the library?:
https://github.com/AmbientDataInc/Ambient_ESP8266_lib

TakehikoShimojima, 2019-06-20:
Yes, it is. It is a library to send IoT data from Arduino sketch to a cloud server "Ambient". https://ambidata.io

cmaglie, 2019-06-25:
@TakehikoShimojima sorry for the delay: you must create a git tag to release your library.
Also the name field does contains a slash `/` that is not permitted: `Ambient ESP32/ESP8266 lib`

To fix this:
1) remove the `/` from the `name` field in library properties
2) commit and push
3) create a git tag called `v0.0.2` and push it

TakehikoShimojima, 2019-06-25:
@cmaglie thank you for your comments.
I removed the `/` from the `name` field, commit and push, created a git tag and push it.

cmaglie, 2019-06-26:
done

closed 2019-06-26 by cmaglie