arduino/Arduino #9004: ESP8266 ESP.reset() not working

DMike92, 2019-06-22:
Hope I'm in the right place...!
Hi,
I am recently using IDE 1.8.9 an discover that ESP.reset is no longer working anymore on ESP8266
Unfortunately, I did not note down the editor version I was using (1.6.x maybe).

I had lot of issues with ESP8266 libraries 2.5.2  and I went back to the last 2.4 (which is 2.4.2) so the problem does not seem to come from the lib.
Real Device: Wemos D1 R2
In IDE:
Board: Wemos D1 R1 or LOLIN(WEMOS) D1 R2 (tried both)
Other settings left as they came (default values)
Prog: AVRISP mkII

matthijskooijman, 2019-06-22:
> Hope I'm in the right place...!

Unfortunately, you're not :-)

You should probably report this at the ESP8266 core you are using (I'm not sure which one you installed, possibly https://github.com/esp8266/Arduino ?). If it turns out this is a problem in the IDE rather than the ESP8266 core, then you should come back here, but I doubt that :-)

closed 2019-06-22 by matthijskooijman

DMike92, 2019-06-22:
Thanks for answering and guiding me.
Have a nice day.