arduino/Arduino #9007: jetson tx2 (aarch64) unable to upload code to arduino

swayam01, 2019-06-23:
Hello,
I am building a robot where I need to communicate from jetson tx2 to arduino using serial port. I have compiled the arduino ide for aarch64 with openjdk-8-jre java_sdk. But unable to upload code to the arduino

error pasted below:
ioctl("TIOCMSET"): Connection timed out
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

P.S: followed issue mentioned here,but no help - https://github.com/arduino/Arduino/issues/7302

Help Appriciated.

swayam01, 2019-06-24:
Issue fixed, serial communication working from jetson to arduino

closed 2019-06-24 by per1234

awesomebytes, 2021-03-24:
@swayam01 how did you fix it? I'm on the same situation (with a different target to upload than an arduino, but it's also a tx2 which communications runs normally, but trying to upload anything to any board always fails with not getting the 'sync' byte.

tonylam1211, 2021-04-18:
Any one may help?? I am also facing the same issue...

@awesomebytes @swayam01