arduino/Arduino #9008: Arduino Nano not uploading sketches: No error information

SeppPenner, 2019-06-23:
I recently fell over the issue that the Arduino Nano was not uploading the sketch (without providing error information).

SeppPenner, 2019-06-23:
The issue can be resolved easily. Go to `Tools` --> `Processor` and set the bootloader to `ATmega328P (Old Bootloader)`. This should be documented in https://www.arduino.cc/en/Guide/Troubleshooting#toc1 under `Bootloader`:

Check that it's not documented:
[image]

SeppPenner, 2019-06-24:
@per1234 Can I contribute on this somehow?

per1234, 2019-06-24:
You can provide some text that you recommend be added to the Troubleshooting page.

There is some existing text about this in the Getting Started page for the Nano that you might be able to work from:
https://www.arduino.cc/en/Guide/ArduinoNano
>NOTE: We have updated the NANO board with a fresh bootloader. Boards sold from us from January 2018 have this new bootloader, while boards manufactured before that date have the old bootloader. First, make sure you have the Arduino AVR Core 1.16.21 or later looking at the Board Manager. Then, to program the NEW Arduino NANO boards you need to chose Processor > "ATmega328P". To program old boards you need to choose Processor > "ATmega328P (Old Bootloader)". If you get an error while uploading or you are not sure which bootloader you have, try each type of processor 328P until your board gets properly programmed.

SeppPenner, 2019-06-24:
To be honest, I would add this text nearls as it is. It seems to describe the error properly. Small changes only:

> Make sure that you're using the correct bootloader when working with the Arduino NANO. Boards sold from us from January 2018 have the new bootloader, while boards manufactured before that date have the old bootloader. First, make sure you have the Arduino AVR Core 1.16.21 or later looking at the Board Manager. Then, to program the NEW Arduino NANO boards you need to chose Processor > "ATmega328P". To program old boards you need to choose Processor > "ATmega328P (Old Bootloader)". If you get an error while uploading or you are not sure which bootloader you have, try each type of processor 328P until your board gets properly programmed.

per1234, 2021-04-04:
https://www.arduino.cc/en/Guide/Troubleshooting has been replaced with [the Arduino Help Center](https://support.arduino.cc/hc), which now has an article with the requested information:
https://support.arduino.cc/hc/en-us/articles/360016785580-Error-avrdude-stk500-getsync-in-Nano-boards
Thanks so much for your excellent suggestion @SeppPenner!

closed 2021-04-04 by per1234

SeppPenner, 2021-04-06:
@per1234 Nice :)