arduino/Arduino #9009: arduino connection/upload problem

liesdotcom, 2019-06-23:
I have a program written in the online arduino editor, but i can't upload anything to the arduino, it does't even give me the option to do so. this is a project for my robotics club, and needs to be done soon, so if i can get help ASAP, that would be great! thank you!

my email is brechinpipeline@gmail.com

alexbertis, 2019-06-23:
Could you elaborate on your problem? Have you checked all connections?

per1234, 2019-06-23:
According to https://github.com/arduino/arduino-create/issues/1#issuecomment-351344954, the [Create > Editor section of the Arduino Forum](http://forum.arduino.cc/index.php?board=101.0) is the appropriate place to submit bug reports or feature requests related to the Arduino Web Editor.

closed 2019-06-23 by per1234