arduino/Arduino #9010: Arduino IDE enhancement request

PickyBiker, 2019-06-23:
The error messages that display in the bottom of the Arduino IDE are currently fixed at a red/orange color for the text and black for the background. 
On my system that is difficult to read. 
Please add a feature to allow adjusting those colors in the preferences.txt file.

per1234, 2019-06-23:
Closing as duplicate of https://github.com/arduino/Arduino/issues/6454.

Note that it is possible to change these colors by editing the theme. You'll find documentation here:
https://github.com/per1234/ThemeTest

per1234, 2019-06-23:
Closing as duplicate of https://github.com/arduino/Arduino/issues/6454.

Note that it is possible to change these colors by editing the IDE's theme. You'll find documentation here:
https://github.com/per1234/ThemeTest

closed 2019-06-23 by per1234